ArduPilot/ardupilot #3586: SITL: Windows build fails (Copter)

shadow1runner, 2016-02-03:
As already mentioned [here](http://ardupilot.com/forum/viewtopic.php?f=71&t=14833) and [here](http://ardupilot.com/forum/viewtopic.php?t=14876&p=40873) there is a problem with building SITL under Windows, under a fresh Cygwin installation with dependencies acc. to the [Wiki](dev.ardupilot.com/wiki/sitl-native-on-windows/).

Removing `-j4` as suggested by @rmackay9 does not change the problem.
A clean rebuild does not help either.

For the sake of completeness here's a complete build output:
```
$ sim_vehicle.sh --map
/cygdrive/e/m/ardupilot/ArduCopter /cygdrive/e/m/ardupilot/ArduCopter
Building sitl
Checking modules
// BUILDROOT=/tmp/ArduCopter.build HAL_BOARD=HAL_BOARD_SITL HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="4c0f0fb8"
echo "Generating MAVLink headers..."
Generating MAVLink headers...
#goto mavlink module directory and run the most recent generator script
echo "Generating C code using mavgen.py located at" /cygdrive/e/m/ardupilot/modules/mavlink/
Generating C code using mavgen.py located at /cygdrive/e/m/ardupilot/modules/mavlink/
PYTHONPATH=/cygdrive/e/m/ardupilot/modules/mavlink/ python /cygdrive/e/m/ardupilot/modules/mavlink//pymavlink/tools/mavgen.py --lang=C --wire-protocol=1.0 --output=/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0 /cygdrive/e/m/ardupilot/modules/mavlink//message_definitions/v1.0/ardupilotmega.xml
Validating /cygdrive/e/m/ardupilot/modules/mavlink//message_definitions/v1.0/ardupilotmega.xml
Parsing /cygdrive/e/m/ardupilot/modules/mavlink//message_definitions/v1.0/ardupilotmega.xml
Note: message DATA64 is longer than 64 bytes long (74 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message DATA96 is longer than 64 bytes long (106 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message REMOTE_LOG_DATA_BLOCK is longer than 64 bytes long (214 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Validating /cygdrive/e/m/ardupilot/modules/mavlink//message_definitions/v1.0/common.xml
Parsing /cygdrive/e/m/ardupilot/modules/mavlink//message_definitions/v1.0/common.xml
Note: message GPS_STATUS is longer than 64 bytes long (109 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ATTITUDE_QUATERNION_COV is longer than 64 bytes long (76 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GLOBAL_POSITION_INT_COV is longer than 64 bytes long (193 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOCAL_POSITION_NED_COV is longer than 64 bytes long (237 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIGHRES_IMU is longer than 64 bytes long (70 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_SENSOR is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SIM_STATE is longer than 64 bytes long (92 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message FILE_TRANSFER_PROTOCOL is longer than 64 bytes long (262 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_STATE_QUATERNION is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOG_DATA is longer than 64 bytes long (105 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GPS_INJECT_DATA is longer than 64 bytes long (121 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SERIAL_CONTROL is longer than 64 bytes long (87 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ENCAPSULATED_DATA is longer than 64 bytes long (263 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message RESOURCE_REQUEST is longer than 64 bytes long (251 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message CONTROL_SYSTEM_STATE is longer than 64 bytes long (108 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message AUTOPILOT_VERSION is longer than 64 bytes long (68 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message V2_EXTENSION is longer than 64 bytes long (262 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Merged enum MAV_CMD
Found 176 MAVLink message types in 2 XML files
Generating C implementation in directory /tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega
Generating C implementation in directory /tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/common
Copying fixed headers
E:/m/ardupilot/mk/mavgen.mk:9: recipe for target '/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/mavlink.h' failed
make: *** [/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/mavlink.h] Aborted (core dumped)
make: *** Deleting file '/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/mavlink.h'
Checking modules
// BUILDROOT=/tmp/ArduCopter.build HAL_BOARD=HAL_BOARD_SITL HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="4c0f0fb8"
echo "Generating MAVLink headers..."
Generating MAVLink headers...
#goto mavlink module directory and run the most recent generator script
echo "Generating C code using mavgen.py located at" /cygdrive/e/m/ardupilot/modules/mavlink/
Generating C code using mavgen.py located at /cygdrive/e/m/ardupilot/modules/mavlink/
PYTHONPATH=/cygdrive/e/m/ardupilot/modules/mavlink/ python /cygdrive/e/m/ardupilot/modules/mavlink//pymavlink/tools/mavgen.py --lang=C --wire-protocol=1.0 --output=/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0 /cygdrive/e/m/ardupilot/modules/mavlink//message_definitions/v1.0/ardupilotmega.xml
Validating /cygdrive/e/m/ardupilot/modules/mavlink//message_definitions/v1.0/ardupilotmega.xml
Parsing /cygdrive/e/m/ardupilot/modules/mavlink//message_definitions/v1.0/ardupilotmega.xml
Note: message DATA64 is longer than 64 bytes long (74 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message DATA96 is longer than 64 bytes long (106 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message REMOTE_LOG_DATA_BLOCK is longer than 64 bytes long (214 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Validating /cygdrive/e/m/ardupilot/modules/mavlink//message_definitions/v1.0/common.xml
Parsing /cygdrive/e/m/ardupilot/modules/mavlink//message_definitions/v1.0/common.xml
Note: message GPS_STATUS is longer than 64 bytes long (109 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ATTITUDE_QUATERNION_COV is longer than 64 bytes long (76 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GLOBAL_POSITION_INT_COV is longer than 64 bytes long (193 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOCAL_POSITION_NED_COV is longer than 64 bytes long (237 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIGHRES_IMU is longer than 64 bytes long (70 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
[... cut ...]
Generating C implementation in directory /tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/common
Copying fixed headers
E:/m/ardupilot/mk/mavgen.mk:9: recipe for target '/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/mavlink.h' failed
make: *** [/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/mavlink.h] Aborted (core dumped)
make: *** Deleting file '/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/mavlink.h'
/cygdrive/e/m/ardupilot/Tools/autotest/sim_vehicle.sh: Build failed
```

Any ideas on this?

shadow1runner, 2016-02-03:
I just noticed that - after the failing build - a python2.7.exe.stackdump is included, it looks as follows:

The python stackdump looks as follows:
```
Stack trace:
Frame     Function  Args
0060C838  61033A23 (00000250, 0000EA60, 000000A4, 0060C8A8)
0060C968  610F27E2 (7B000279, FFFFFFFE, 2D49E032, 0092D778)
```

This made me curious and then I noticed that the cygwin installation comes with a separate python version, I removed this version from PATH (and thus had a fallback to Anaconda 2.7 x64 Python). This changed the whole picture:

```
$ make sitl -j4
echo "Generating MAVLink headers..."
Checking modules
Generating MAVLink headers...
#goto mavlink module directory and run the most recent generator script
echo "Generating C code using mavgen.py located at" /cygdrive/e/m/ardupilot/modules/mavlink/
Generating C code using mavgen.py located at /cygdrive/e/m/ardupilot/modules/mavlink/
PYTHONPATH=/cygdrive/e/m/ardupilot/modules/mavlink/ python /cygdrive/e/m/ardupilot/modules/mavlink//pymavlink/tools/mavgen.py --lang=C --wire-protocol=1.0 --output=/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0 /cygdrive/e/m/ardupilot/modules/mavlink//message_definitions/v1.0/ardupilotmega.xml
C:\Anaconda2\python.exe: can't open file '/cygdrive/e/m/ardupilot/modules/mavlink//pymavlink/tools/mavgen.py': [Errno 2] No such file or directory
E:/m/ardupilot/mk/mavgen.mk:9: recipe for target '/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/mavlink.h' failed
make: *** [/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/mavlink.h] Error 2
make: *** Waiting for unfinished jobs....
// BUILDROOT=/tmp/ArduCopter.build HAL_BOARD=HAL_BOARD_SITL HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="4c0f0fb8"
```
I checked
'/cygdrive/e/m/ardupilot/modules/mavlink//pymavlink/tools/mavgen.py': [Errno 2] No such file or directory`, procmon showing the following:

[image]

Notice the path that python tries to access `E:\cygdrive\e\m\ardupilot\modules\mavlink\pymavlink\tools\mavgen.py` does _not_ exist, the correct path is `E:\m\ardupilot\modules\mavlink\pymavlink\tools\mavgen.py`.

Am I mistaken or wasn't the Wiki suggesting MinGw for building SITL up to now?
Cause so far there seem to be two problems:

  * incompatible python version that gets installed (and is used by default) in cygwin
  * wrong path building (how to fix?)

meee1, 2016-02-03:
build started failing around the 15-1-2016 according to my sitl image generator.

guludo, 2016-02-04:
@shadow1runner @meee1 

Hi guys, I think I know the problem. Probably because the mavlink submodule isn't on your local tree. I think this problem started after https://github.com/diydrones/ardupilot/pull/3504 .

If you do:

```bash
git submodule init
git submodule update
```

Then, probably the problem is solved.

Hm, I think we should change the build system to init and update submodule dependencies before.

craigo-, 2016-02-04:
Thanks guludo... I remember previously trying "git submodule update" after the failure (because the [submodules url]( http://dev.ardupilot.com/wiki/git-submodules/) mentioned in the command output also suggested doing this), but it didn't fix the problem then.

I have retried the process completely afresh on a different computer. The output is the same as I posted in the forum post "[Major problem when installing SITL on Windows](http://ardupilot.com/forum/viewtopic.php?f=71&t=14833#p40804)", including the errors. The output appears to call "git submodule init" (line 7) and "git submodule update" (line 16). As per your suggestion, I then tried executing "git submodule init" and "git submodule update" from the ~/ardupilot directory. The commands both completed successfully with no output. Rerunning the make command from the ArduCopter directory results in the same errors as before (documented below for completeness).

It looks like rerunning the git submodules commands does not solve the problem. I checked the ~/ardupilot/modules/ directory and there is definitely a mavlink directory therein.

```
user@computer ~/ardupilot/ArduCopter
$ make sitl -j4
echo "Generating MAVLink headers..."
Checking modules
Generating MAVLink headers...
#goto mavlink module directory and run the most recent generator script
echo "Generating C code using mavgen.py located at" /home/user/ardupilot/modules/mavlink/
Generating C code using mavgen.py located at /home/user/ardupilot/modules/mavlink/
PYTHONPATH=/home/user/ardupilot/modules/mavlink/ python /home/user/ardupilot/modules/mavlink//pymavlink/tools/mavgen.py --lang=C --wire-protocol=1.0 --output=/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0 /home/user/ardupilot/modules/mavlink//message_definitions/v1.0/ardupilotmega.xml
Validating /home/user/ardupilot/modules/mavlink//message_definitions/v1.0/ardupilotmega.xml
Parsing /home/user/ardupilot/modules/mavlink//message_definitions/v1.0/ardupilotmega.xml
// BUILDROOT=/tmp/ArduCopter.build HAL_BOARD=HAL_BOARD_SITL HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="a1b0a53e"
Note: message DATA64 is longer than 64 bytes long (74 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message DATA96 is longer than 64 bytes long (106 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message REMOTE_LOG_DATA_BLOCK is longer than 64 bytes long (214 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Validating /home/user/ardupilot/modules/mavlink//message_definitions/v1.0/common.xml
Parsing /home/user/ardupilot/modules/mavlink//message_definitions/v1.0/common.xml
Note: message GPS_STATUS is longer than 64 bytes long (109 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ATTITUDE_QUATERNION_COV is longer than 64 bytes long (76 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GLOBAL_POSITION_INT_COV is longer than 64 bytes long (193 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOCAL_POSITION_NED_COV is longer than 64 bytes long (237 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIGHRES_IMU is longer than 64 bytes long (70 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_SENSOR is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SIM_STATE is longer than 64 bytes long (92 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message FILE_TRANSFER_PROTOCOL is longer than 64 bytes long (262 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_STATE_QUATERNION is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOG_DATA is longer than 64 bytes long (105 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GPS_INJECT_DATA is longer than 64 bytes long (121 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SERIAL_CONTROL is longer than 64 bytes long (87 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ENCAPSULATED_DATA is longer than 64 bytes long (263 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message RESOURCE_REQUEST is longer than 64 bytes long (251 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message CONTROL_SYSTEM_STATE is longer than 64 bytes long (108 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
[... cut ...]
Generating C implementation in directory /tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/common
Copying fixed headers
C:/cygwin/home/user/ardupilot/mk/mavgen.mk:9: recipe for target '/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/mavlink.h' failed
make: *** [/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/mavlink.h] Aborted (core dumped)
make: *** Deleting file '/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/mavlink.h'
make: *** Waiting for unfinished jobs....
```

shadow1runner, 2016-02-05:
Hi guys,

sry for the delay. I've tried @guludo 's suggestion as well, but with the same outcome as @craigo- . I've tried the very same on a different windows machine (cloning `master` from scratch) - same outcome.

meee1, 2016-02-06:
workarounds.
edit mavgen.mk line 12, and add a "-" to the start of the line
or
make sitl -I

python is crashing during the mavlink headers. but is actually completing.

craigo-, 2016-02-06:
Thank you, @meee1.

I can confirm that SITL successfully compiles using both workarounds:

- Workaround 1: Small clarification ... I found that the dash must be placed _after_ the tab and immediately _before_ the word "PYTHONPATH".
- Workaround 2: `make sitl -i` completes successfully. (I first tried `make sitl -j4 -i`, which core-dumped.)

I have a working SITL again - yay! I hope this points the team in the right direction for a more permanent fix. Very happy to retest as required and provide feedback.

magicrub, 2016-02-16:
fixed via https://github.com/diydrones/ardupilot/commit/7fda3ba17843c1fbb331c5c600571171d9da2421

craigo-, 2016-02-16:
Thanks, @magicrub. Your commit looks like it enacts the first workaround suggested by @meee1. Are there plans to do anything further to the ardupilot codebase to solve the underlying problem, if indeed the problem lies somewhere in ardupilot? Or is the problem somewhere upstream, e.g. in cygwin or python?

nocternal, 2016-02-16:
when I enter sim_vehicle.sh -j4 --console --map
the result is 
Window access not found, logging to /tmp/ardupilot.log
but I have proxy pompt ,aonsole and map
but can't load parameters
when enter  
param load ..\Tools\autotest\ArduPlane.cpp
the result is
Failed to open file '..\Tools\autotest\ArduPlane.cpp'
any ideas to solve it?

craigo-, 2016-02-16:
@nocternal, firstly that's an odd extension for a parameters file. For copter, I'm loading a file named "copter_params.parm". Looking at my Tools/autotest directory, there's a likely-looking file for you named "ArduPlane.parm".

Secondly, try using forward slashes. I found that at some point, possibly after a MAVProxy update, that backslashes no longer worked.

So, instead try `param load ../Tools/autotest/ArduPlane.parm`

(I am on a Windows platform, by the way.)

nocternal, 2016-02-16:
@craigo-   I try as you said.It works! Thanks a lot! ^_^

lucasdemarchi, 2016-02-16:
@magicrub adding a `-` will just make make ignore the return value. I don't think it's the proper solution.... If the mavgen tool fails, we actually don't want to proceed.... the fix should be inside the tool so it works correctly on windows.

magicrub, 2016-02-16:
I'm open to suggetsions

lucasdemarchi, 2016-02-16:
@magicrub from my part I'll postpone any investigation on this after waf is building ardupilot on windows.  Just bear in mind it's not actually fixing the issue.

magicrub, 2016-02-16:
understood.

galoulebougalou, 2016-04-27:
Hi everybody and Help !!
I tried to follow all steps but neither make sitl -j4 neither make sitl -i seem to work with me.
Checked .bashrc - checked mavgen.mk and tried to run with both of the above make comands.
Tried also git submodule init and git submodule update
Any idea ?

`make sitl -i
Checking modules
// BUILDROOT=/tmp/ArduCopter.build HAL_BOARD=HAL_BOARD_SITL HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="e83b10cb" -I/home/Fab/ardupilot/libraries/AP_Common/missing -DHAVE_CMATH_ISFINITE -DNEED_CMATH_ISFINITE_STD_NAMESPACE
%% compat.o
In file included from /home/Fab/ardupilot/libraries/AP_Math/AP_Math.h:17:0,
                 from /home/Fab/ardupilot/libraries/GCS_MAVLink/GCS_MAVLink.h:9,
                 from /home/Fab/ardupilot/libraries/GCS_MAVLink/GCS.h:10,
                 from Copter.h:42,
                 from compat.cpp:1:
/home/Fab/ardupilot/libraries/AP_Math/quaternion.h: In member function 'bool Quaternion::is_nan() const':
/home/Fab/ardupilot/libraries/AP_Math/quaternion.h:51:24: error: 'isnan' was not declared in this scope
         return isnan(q1) || isnan(q2) || isnan(q3) || isnan(q4);
                        ^
compilation terminated due to -Wfatal-errors.
C:/cygwin/home/Fab/ardupilot/mk/build_rules.mk:11 : la recette pour la cible « /tmp/ArduCopter.build/compat.o » a échouée
make: [/tmp/ArduCopter.build/compat.o] Erreur 1 (ignorée)
%% ArduCopter.o
In file included from /home/Fab/ardupilot/libraries/AP_Math/AP_Math.h:17:0,
                 from /home/Fab/ardupilot/libraries/GCS_MAVLink/GCS_MAVLink.h:9,
                 from /home/Fab/ardupilot/libraries/GCS_MAVLink/GCS.h:10,
                 from Copter.h:42,
                 from ArduCopter.cpp:76:
/home/Fab/ardupilot/libraries/AP_Math/quaternion.h: In member function 'bool Quaternion::is_nan() const':
/home/Fab/ardupilot/libraries/AP_Math/quaternion.h:51:24: error: 'isnan' was not declared in this scope
         return isnan(q1) || isnan(q2) || isnan(q3) || isnan(q4);
                        ^
compilation terminated due to -Wfatal-errors.
C:/cygwin/home/Fab/ardupilot/mk/build_rules.mk:11 : la recette pour la cible « /tmp/ArduCopter.build/ArduCopter.o » a échouée
make: [/tmp/ArduCopter.build/ArduCopter.o] Erreur 1 (ignorée)
%% control_drift.o
In file included from /home/Fab/ardupilot/libraries/AP_Math/AP_Math.h:17:0,
                 from /home/Fab/ardupilot/libraries/GCS_MAVLink/GCS_MAVLink.h:9,
                 from /home/Fab/ardupilot/libraries/GCS_MAVLink/GCS.h:10,
                 from Copter.h:42,
                 from control_drift.cpp:3:
/home/Fab/ardupilot/libraries/AP_Math/quaternion.h: In member function 'bool Quaternion::is_nan() const':
/home/Fab/ardupilot/libraries/AP_Math/quaternion.h:51:24: error: 'isnan' was not declared in this scope
         return isnan(q1) || isnan(q2) || isnan(q3) || isnan(q4);
                        ^
compilation terminated due to -Wfatal-errors.
C:/cygwin/home/Fab/ardupilot/mk/build_rules.mk:11 : la recette pour la cible « /tmp/ArduCopter.build/control_drift.o » a échouée
make: [/tmp/ArduCopter.build/control_drift.o] Erreur 1 (ignorée)
%% control_circle.o
In file included from /home/Fab/ardupilot/libraries/AP_Math/AP_Math.h:17:0,
                 from /home/Fab/ardupilot/libraries/GCS_MAVLink/GCS_MAVLink.h:9,
                 from /home/Fab/ardupilot/libraries/GCS_MAVLink/GCS.h:10,
                 from Copter.h:42,
                 from control_circle.cpp:3:
`
.........

`C:/cygwin/home/Fab/ardupilot/mk/board_native.mk:124 : la recette pour la cible « /tmp/ArduCopter.build/ArduCopter.elf » a échouée
make: [/tmp/ArduCopter.build/ArduCopter.elf] Erreur 1 (ignorée)
cp: cannot stat '/tmp/ArduCopter.build/ArduCopter.elf': No such file or directory
C:/cygwin/home/Fab/ardupilot/mk/board_native.mk:124 : la recette pour la cible « /tmp/ArduCopter.build/ArduCopter.elf » a échouée
make: [/tmp/ArduCopter.build/ArduCopter.elf] Erreur 1 (ignorée)
Firmware is in ArduCopter.elf
`

lucasdemarchi, 2016-04-27:
You have a recent version of the toolchain which broke the build. You can build it using waf: see https://github.com/ArduPilot/ardupilot/pull/3878  - it's not integrated with sim_vehicle.sh yet, though (one of the reasons we didn't switch to it yet). Patches to do that are welcome.

pjmosquera, 2016-04-27:
@lucasdemarchi hello Lucas. I don`t get it #3878 Which steps for dummies I must to follow on Windows. I have all recent versions recomended on http://ardupilot.org/dev/docs/sitl-native-on-windows.html#sitl-native-on-windows.  For several days my compilation terminated due to -Wfatal-errors.
Thank you

galoulebougalou, 2016-04-27:
@pjmosquera 
In mk directory you have board_native.mk file. Edit it and in line 6, replace :
 `EXTRAFLAGS += -DHAVE_CMATH_ISFINITE -DNEED_CMATH_ISFINITE_STD_NAMESPACE  `with  `EXTRAFLAGS += -DHAVE_CMATH_ISFINITE -DNEED_CMATH_ISFINITE_STD_NAMESPACE  -DHAVE_CMATH_ISINF -DNEED_CMATH_ISINF_STD_NAMESPACE -DHAVE_CMATH_ISNAN -DNEED_CMATH_ISNAN_STD_NAMESPACE`
You should be able to run 
` make sitl -j4`

pjmosquera, 2016-04-27:
OMG Unbelievable! "Firmware is in ArduCopter.elf"
Thank you very muck @galoulebougalou . This is it

OXINARF, 2016-05-03:
Closing this in favor of #4018

closed 2016-05-03 by OXINARF