ArduPilot/ardupilot #3587: Error in Lightware Range Finder Serial Driver (Easy fix!)

heinduplessis, 2016-02-04:
The Lightware Serial driver is erroneously sending a Line Feed character ("/n") to request a range reading, in stead of the letter "d".

This is no problem on the SF/10 and higher models which will send the distance by default, regardless of the character, however the SF/02 does not respond to anything else than the "d".

I've checked this with James from Lightware and he confirms it.

The fix is simple, in library:

 ardupilot/libraries/AP_RangeFinder/AP_RangeFinder_LightWareSerial.cpp

Line 77 & 78, the source reads:

    // we need to write a byte to prompt another reading
    uart->write('\n');

The last line just needs to change to:
    uart->write('d');

Can somebody compile a custom bin for the Pixhawk plane so that I can test? I will confirm the fix. Alternatively I can pull and compile, but we need it in the main code branch.

Many thanks
Hein

magicrub, 2016-02-14:
@tridge confirms that this change is most likely correct.

magicrub, 2016-02-15:
Fixed! https://github.com/diydrones/ardupilot/commit/786d14cd3cad777fa0cab5dc4338a8c4f2b20da7 Thanks @heinduplessis for the bug report and for pointing to the problem, twas much easier that way.

closed 2016-02-15 by magicrub