ArduPilot/ardupilot #3598: Builds failing without explicit submodule init/update

hamishwillee, 2016-02-08:
I've just updated the [PX4 build instructions](http://dev.ardupilot.com/wiki/building-px4-for-linux-with-make/) to add explicit submodule intit. This is something that the toolchain should do for us.

guludo, 2016-02-11:
The waf-based build system was added with that feature recently, i.e., submodule checks on build. The PX4 build in waf is still a work in progress though, I think we'll have it soon.

OXINARF, 2017-06-02:
waf build system now updates submodules by default (with an option to turn it off) if the current commit is older than should be.
Make build system inits submodules but doesn't update them.

I think this means this issue is fixed now. Feel free to re-open if you think otherwise.

closed 2017-06-02 by OXINARF

WickedShell, 2017-06-02:
@OXINARF yeah is there a flag that turns that off in WAF? I actually find that to be a problem when working on changes to a submodule that I want to test in SITL first... (Not to hijack the issue)

guludo, 2017-06-02:
@WickedShell Is `waf configure --no-submodule-update` what you're looking for?

WickedShell, 2017-06-02:
@guludo that looks exactly like it from the --help! Thanks :)

OXINARF, 2017-06-02:
@WickedShell So it is updating your submodule when you are working on them? I think it shouldn't - if you can get a reproducible case when this happens we can probably fix it.

WickedShell, 2017-06-02:
@OXINARF I might be confusing which tool it was at this point. When I was trying to work on the accel cal stuff the build system was trying to get the correct MAVLink commit checked out, but was failing as I had modified files in the submodule. I ended up just making a commit, and then committing it to the ardupilot repo so that the build would work.