ArduPilot/ardupilot #3600: Plane: remove plane ALT_OFFSET?

magicrub, 2016-02-08:
Why do we have Alt Offset? It does the same thing as GND_ALT_OFFSET, right? It seems to work like terrain alt but is therefore redundant. I vote we remove it from plane.

Do you use ALT_OFFSET? Can you tell us when you use it?

ALT_OFFSET
http://plane.ardupilot.com/wiki/arduplane-parameters/#altitude_offset_arduplanealt_offset

GND_ALT_OFFSET
http://plane.ardupilot.com/wiki/arduplane-parameters/#altitude_offset_gnd_alt_offset

WickedShell, 2016-03-15:
One of the reasons I can think of for having the non GND_ one is if you are flying GPS altitude and want to adjust the offset for some reason. That use case seems kinda rare though, as I can't think of why you would need to adjust the mission in this way.

On the other hand GND_ALT_OFFSET is available to all platforms. I think the ALT_OFFSET variant could be safely removed unless @tridge has a use case that means we should keep it around.

magicrub, 2016-05-27:
any thoughts on this (removing param ALT_OFFSET)?

tridge, 2016-06-04:
they aren't the same thing, ALT_OFFSET works on the mission, GND_ALT_OFFSET on the sensor (baro)
I use ALT_OFFSET a lot, so I'd like to keep it :-)

closed 2016-06-04 by tridge