ArduPilot/ardupilot #3601: Plane: add deadband pause to throttle slew when throttle switches directions

magicrub, 2016-02-09:
Add deadband pause to throttle slew when throttle switches directions. This helps for reverse thrust capable ESCs designed for cars where going forward to reverse requires a pause at neutral. Suggest adding new param for how long you want the delay to be. Possible param name THR_SLEW_Z_PAUSE?

iskess, 2016-02-23:
+1. Reversing the motor direction with a large propeller could create undesirable mechanical and aerodynamic effects. Unfortunately the BLHeli firmware frequently used to allow for reverse thrust on ESCs meant for aircraft doesn't not provide a "Brake" to stop the prop when throttle is removed. A large windmilling prop will further exacerbate the problem. Having a parameter to increase the transition time will be helpful. 
Does Thottle_Slew apply to the reverse range? I propose that the reverse throttle range should have its own slew parameter. I would select a much slower slew for reverse thrust than forward.

Naterater, 2019-01-04:
I really like this idea. When a plane is on a perfect glideslope, the throttle will often oscillate between forward and reverse quickly. I can't imagine this is good for ESC's.

IamPete1, 2021-09-12:
https://github.com/ArduPilot/ardupilot/pull/18099