ArduPilot/ardupilot #3605: Pixracer: SBUS not working with Copter-3.4-dev

rmackay9, 2016-02-10:


olexs, 2016-02-17:
The 3.3.3 beta release available through the Mission Planner has the same problem. It definitely used to work in an earlier build of 3.4dev.

MR444, 2016-02-19:
Just tested SBUS (X4R) and 3.4-dev tricopter. No problems!
[image]

rmackay9, 2016-02-20:
@MR444, great, thanks for testing.

Fixed in Copter-3.3.4-rc2.

closed 2016-02-20 by rmackay9

govsux, 2016-03-19:
FYI Cannot get my X8R Frysky receiver to work on the P racer sbus. It doesn't see it. I'm running the PX4 flight stack so it seems it's more than just amp copter code???

olexs, 2016-03-19:
SBUS works fine with PX4 stack for me. Apparently, it's also fixed in APM now - I haven't tested, but have seen several videos of people using X4R-SB receivers with the Pixracer and APM. Will confirm tomorrow, since I'm planning on reflashing to APM anyway.

lorbass, 2016-03-21:
@govsux 
Same issue with Pixracer. PX4 (master, Beta, stable) do NOT recognize the radio channels.
APM (3.3.3) flashed with QGC or MP the radio channels are reconized and no problem to calibrate.

govsux, 2016-03-21:
@Lorbass  Yes same here. Also APM 3.4 dev doesn't see bus RX.  3.3.3 seems fine. Except... the wifi only works with PX4 not APM. Guess you just can't have your cake and eat it too yet.

govsux, 2016-03-21:
APM 3.4 sbus does work now...

lorbass, 2016-03-22:
@govsux I do not like Wifi because of possible interference with the 2.4 GHz TX/RX.
e.g. strongly recommended to disable Wifi at the Gopro camera. But I would prefer PX4 do to
the possiblity of VTOL Tailsitter.
I wonder why the PX4 developer not react on my Issue message in the appropriate List.
APM Developer are capable to solve the issue on 3.4 in no time

govsux, 2016-03-22:
@lorbass  Yes that does seem like it could create interference but there is already 2.4 mhz flying all over the place and no issue. But being so close to the RX does seem sort of risky. I would love it if we had an option in the params to shut wifi off when the board arms. Is that the VTOL Tailsitter you refer too? I read somewhere that has been discussed already. Re- PX4 devs... I think they are just really busy sorting out minor issues and growing pains. I did find out 3dr radio works fine with APM 3.3.3 and 3.4. It's odd that works but not blue tooth. Maybe it's my GCS or something. But as long as the 3dr radio's work I'm good to go. Anyway... as you know there are these discussions here too http://diydrones.com/forum/topics/pixracer-with-ardupilot?

rezafarokhian, 2016-10-16:
hi. please help me. my futaba 8fg does not connecting s.bus to pixracer no channel detected in qgroundcontrol

lorbass, 2016-10-16:
Nobody can help with no more detailed information.
e.g. which versions for QG and Firmware.
And why QG in order to use Ardupilot Firmware and not Missionplanner?
http://ardupilot.org/planner/docs/common-install-mission-planner.html
For me it's like a VW with Mercedes Engine.
Pixracer is something special. It works from AC 3.4 only.
In Missionplanner you will have to select Beta Version AC 3.4 rc5 so far.

lvale, 2016-10-16:
3.4 RC6 :) ;)

Biswajit1995, 2017-04-25:
Same Here,

I have upgraded my Pixhawk firmware from copter 3.3.3 to copter 3.4 and, in RC calibration none of the channels are responding in QGroundControl and no green bars are showing up in mission planner. Don't know what to do.

Please help, if anyone has solved this issue.

Note: I am using FUTABA T14SG transmitter and FUTABA R7008SB receiver. I have tested my receiver and transmitter at least 50 times, no issue found on them.

clolsonus, 2021-07-02:
This is a really old thread, but I'm playing around with a pixracer here with an frsky sbus receiver.  PX4 detects it fine and I've flown it with PX4 firmware.  Was just seeing what would happen with ardupilot firmware and it cannot detect the sbus receiver reporting 0 channels detected.
When I run the examples/RCInput it also reports zero channels detected.
If I just go ahead and read the channels though, they are there and I get correct values.
Summer: hal.rcin->num_channels() reports zero channels on a pixracer with working and properly connected sbus.  (But you can go ahead and read the channels anyway and they are correct.)  Sadly, the copter firmware depends on num_channels() so you can't setup the radio.  Bummer ... back to px4 I guess for this little f450 experiment.

JunOllyLi, 2022-12-10:
Still seeing the same problem as closonus reported. I did not dig into it but with the same RX and wiring, PX4 FW works but not Ardupilot (4.3.1 FW)

rmackay9, 2022-12-13:
@JunOllyLi,

It's probably better to report in the forums first or raise a new issue but one difference is that AP can accept all RC inputs types on the RC input pins.  So both PWM and SBUS receivers can be connected to the autopilot's RC input pin and it should work.  So.. I think it is a wiring issue on the pixracer you're using.

JunOllyLi, 2022-12-13:
Thanks @rmackay9 ! I was not aware of the existence of the forum. I was just googling the issue I was seeing and found this one.
I am using an RX with only SBUS output, no PWM/PPM. And I connected to the RC-IN of PIXRacer. And with the RX connected, no change in the wire, switching between Ardupilot FW and PX4 FW makes the difference. So I doubt it is a wiring issue.