ArduPilot/ardupilot #3606: PixRacer: bad gyro health message appearing intermittently

rmackay9, 2016-02-10:
This issue appears both with Copter-3.3.3 and Copter-3.4-dev and it's always the first IMU's gyros and accelerometers that appear as unhealthy.

WickedShell, 2016-02-10:
I've been seeing this on a pixhawk as well starting when I shifted to using
EKF2 (plane 3.5 and master). Not sure if it is explicitly related to that
shift as I was using a new pixhawk and new firmware at the same time, but
thought I'd note that I've had this problem recently as well.

ycai47, 2016-02-11:
I have the same issue on a brand pixhawk and newly loaded firmware.

rmackay9, 2016-02-20:
fixed in Copter-3.3.4-rc2.

closed 2016-02-20 by rmackay9

mohitgodiya, 2016-08-22:
Hello,
I am facing the same problem since today.
I have a brand new pixhawk that I bought about a month ago, I waited for all my parts to arrive and before going for my 4th liftoff, it made an annoying sound which stopped after some time. 
When I armed the vehicle and raised the throttle, it started spinning 360deg on the spot.
Next time when the annoying sound came up again, I connect the pixhawk through usb and mission planner said "BAD GYRO HEALTH"....

Please help.
Can upload video and flight logs if you want.

magicrub, 2016-08-22:
Yes, we always want logs.

mohitgodiya, 2016-08-22:
Please find attachment as requested.

[2016-08-21 20-14-40.zip](https://github.com/ArduPilot/ardupilot/files/429574/2016-08-21.20-14-40.zip)

TimothyGold, 2017-04-23:
I just installed a pixracer running 3.4.6 into a new Quadcopter and am finding now on every boot, I initially get a bad Gyro alarm.  It continues to warn me and the message stays on the Mission Planner screen.  The quad eventually passes all the checks and lets me arm and I can fly fine.  Once I arm the message disappears.  I haven't fully tested all the flight modes yet but Alt Hold seems to be fine.  I have the same version of pixracer in a plane running 3.7.1 and I don't get these alarms on the other pixracer.  I don't know if I have a bad board or if this is an indication that this bug re-appeared.  I enabled logging while disarmed so that I could get a log of the status.  I don't see anything in the log to indicate a problem but hopefully someone that knows can take a look and see.
[2017-04-23 15-32-34.zip](https://github.com/ArduPilot/ardupilot/files/950062/2017-04-23.15-32-34.zip)

OXINARF, 2017-04-23:
@TimothyGold We don't do support here, please post in the forum (http://discuss.ardupilot.org).