ArduPilot/ardupilot #3613: Terrain follow - verify height

GiovanniVe, 2016-02-12:
Hi there.
Using Mission Planner (latest beta) to plan flights for an octacopter (Pixhawk with APM 3.3.3 rc2), when  MP and Pixhawk are not connected (neither via USB nor MAVLink) I'm able to set the Terrain follow (from the dropdown menu) and tick the Verify Height option, and they seem to work (as shown in Map Tools - Elevation Graph); however when I connect MP to the Pixhawk, and retrieve a planned mission or just go to Flight Plan window, the "Absolute - Relative - Terrain" dropdown disappears, Verify Height checkbos is there, but am not able to plan a mission at constant height from Terrain (the Elevation Graph shows only a Relative or Absolute height, not a constant height).
Perhaps this is a normal scenario that will be handled properly with future firmware releases.
Any thought or advice is welcome.
Thanks
Giovanni

rmackay9, 2016-02-13:
That's right.  Terrain following is not in Copter-3.3.3.  It planned for Copter-3.4 so there's an issue here for it: https://github.com/diydrones/ardupilot/issues/1305

closed 2016-02-13 by rmackay9

GiovanniVe, 2016-02-13:
All right; that's what I figured; just wanted to be sure.
Thanks a lot.
BR
Giovanni

Date: Fri, 12 Feb 2016 20:09:47 -0800

Closed #3613.