ArduPilot/ardupilot #3619: Add support for MAVLINK_MSG_ID_BATTERY_STATUS

DonLakeFlyer, 2016-02-14:
Currently a custom message MAVLINK_MESSAGE_INFO_BATTERY2 is sent. Can you either switch away from the custom message to the mavlink standard MAVLINK_MSG_ID_BATTERY_STATUS. Or send MAVLINK_MSG_ID_BATTERY_STATUS as well?

Make it difficult for ground stations to handle battery information in a generic way.

WickedShell, 2016-02-14:
Sounds reasonable to me, I'm opposed to all the different _2 variants :)

DonLakeFlyer, 2016-02-15:
Ok, I was wrong about ArduPilot sending MAVLINK_MESSAGE_INFO_BATTERY2. Although it is in the ArduPilot mavlink dialect it doesn't seem to be used. It would still be good to implement MAVLINK_MSG_ID_BATTERY_STATUS though since it provides additional information such as you can derive cell count from it.

DonLakeFlyer, 2016-02-15:
Also since ArduPilot doesn't send MAVLINK_MESSAGE_INFO_BATTERY2 neither MP nor AP2 are using it. So MAVLINK_MSG_ID_BATTERY_STATUS just makes additional information available. Requires no changes to existing GCS implementations (unless they want to support new info).

magicrub, 2016-02-15:
I use battery 2 for debug :)

WickedShell, 2017-04-10:
Implemented in #6015 once thats merged this can be closed.

WickedShell, 2017-04-12:
@DonLakeFlyer  #6015 is in, closing this now.

closed 2017-04-12 by WickedShell