ArduPilot/ardupilot #3620: Improve in-source battery documentation: FS_BATT_ENABLE, FS_BATT_VOLTAGE, FS_BATT_MAH

hamishwillee, 2016-02-14:
The documentation on FS_BATT_ENABLE, FS_BATT_VOLTAGE, FS_BATT_MAH does not make it clear that FS_BATT_VOLTAGE, FS_BATT_MAH depend on the first parameter for the handling in the event of a trigger (do nothing, RTL, LAND) but instead state that it will RTL.

My proposal is the following:

FS_BATT_ENABLE

From

> Controls whether failsafe will be invoked when battery voltage or current runs low.

To

> Controls whether battery failsafes are enabled, and the behaviour when a failsafe is triggered: LAND or RTL. Failsafe trigger levels are set in the parameters FS_BATT_VOLTAGE (battery voltage) and FS_BATT_MAH (battery capacity).

FS_BATT_VOLTAGE

From

> Battery voltage to trigger failsafe. Set to 0 to disable battery voltage failsafe. If the battery voltage drops below this voltage then the copter will RTL

To

> Low battery voltage failsafe trigger level (a value of 0 effectively disables this failsafe). The behaviour when a failsafe is triggered is set in FS_BATT_ENABLE: do nothing, land or RTL.

 FS_BATT_MAH

From

> Battery capacity remaining to trigger failsafe. Set to 0 to disable battery remaining failsafe. If the battery remaining drops below this level then the copter will RTL

To

> Low battery capacity failsafe trigger level (a value of 0 effectively disables this failsafe). The behaviour when a failsafe is triggered is set in FS_BATT_ENABLE: do nothing, land or RTL.

Note that the original proposal (https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/291#issuecomment-183614205) was to include information "If the battery remaining drops below this level then the copter will send a “Low Battery!” message via MavLink to the ground stations HUD (if using telemetry), activate a Loud beeping if a buzzer is attached, flash LEDs and based on the value of FS_BATT_ENABLE either Land, RTL or continue.". I wasn't sure how useful the buzzer/LED stuff was. Feel free to include if it is generic and useful.

WickedShell, 2018-03-10:
@hamishwillee if you have a chance I'd appreciate a pass over the param documentation in #7213. The descriptions need to be valid for all vehicles, while the param values can be specific per vehicle.

hamishwillee, 2018-03-10:
@WickedShell You caught me in a spare moment while marinating beef jerky :-) I think I reviewed them all - it is a big PR. Cheerio

WickedShell, 2018-03-27:
This was fixed as a part of #7213

closed 2018-03-27 by WickedShell