ArduPilot/ardupilot #3621: Waypoint Radius Minimum/Increment in Arducopter/APMRover2

m11s, 2016-02-15:
I have an application that needs the waypoint radius to be 1cm or less for both Arducopter and APMrover2. I have found several parameters linked to this behavior(list below) and was hoping that this could be solved by one or both of the following:
a.) changing the WP_RADIUS Increment from .1m to .001 or .01
b.) changing the RADIUS range from 100cm to .1 or 1

This need for more precision arises from the new RTK/DGPS GNSS receivers that can provide cm level accuracy. Like Piksi and Emlid Reach

I'm not entirely clear if any of the following parameters are platform specific or not?
Param: RADIUS
DisplayName: Waypoint Radius
https://github.com/diydrones/ardupilot/blob/c8661f804f2985e329d8769f2a07e816d13369e3/libraries/AC_WPNav/AC_WPNav.cpp#L19

WP_RADIUS
https://github.com/diydrones/ardupilot/blob/be456fc33dc16e1be7646cb969e0e8b86a58fbd5/APMrover2/Parameters.cpp#L395

_wp_radius_cm
https://github.com/diydrones/ardupilot/blob/4d458833dc73264cd13c95433e33c4c5bcc615f1/libraries/AC_WPNav/AC_WPNav.h#L290

WPNAV_WP_RADIUS
WP_MAX_RADIUS
WPNAV_RADIUS
WAYPOINT_RADIUS

magicrub, 2016-02-15:
The increment value is only a suggestion. It's either integer or float and
if it's float you can put in any value.

magicrub, 2016-02-16:
The accuracy of the waypoint is not that accurate quite yet. We send data via float which resolves to a meter or so. Changing it to transmit waypoints via integer will get us down cm level. It's on the TODO list.

m11s, 2016-02-16:
The accuracy can be achieved by using COMMAND_INT instead of COMMAND_LONG as shown in the following two links.
https://github.com/diydrones/ardupilot/issues/1907
http://mavlink.org/messages/common#COMMAND_INT

That's good if the increment is just a suggestion. 
But, the link below says that the copter only uses [WPNAV_RADIUS](http://copter.ardupilot.com/wiki/arducopter-parameters/#waypoint_radius_wpnav_radius) and no waypoint radius is given for the rover.
http://planner.ardupilot.com/wiki/common-mavlink-mission-command-messages-mav_cmd/#mav_cmd_nav_waypoint
WPNAV_RADIUS is like "RADIUS" from the initial question in that it has range 100-1000 in increments of 1cm. I can not find if WPNAV_RADIUS is an Int./float or if it is the same as WP_RADIUS?

So it looks like we need to lower WPNAV_RADIUS's range minimum to zero like WP_RADIUS is.

m11s, 2016-02-20:
I think I know what's what now. WPNAV_RADIUS is the same as what I refered to as RADIUS. RADIUS is in the library AC_WPNav. So, according to item 30 [here](http://copter.ardupilot.com/wiki/arducopter-parameters/) the names are structured as WPNAV_ Parameters which gives us [WPNAV_RADIUS](https://github.com/diydrones/ardupilot/blob/master/libraries/AC_WPNav/AC_WPNav.cpp#L19).

And, the Range is used by mission planner(GCS) to give **suggested** param values but can be overridden.

closed 2016-02-20 by m11s