ArduPilot/ardupilot #3625: Copter: Use of Lidar in Auto mode

Benbenbeni, 2016-02-16:
Hi everyone.

I'm flying with a PixHawk (v3.4) on a Quadcopter equiped with a Lidar. The latter works fine on Loiter mode however, in Auto mode, I can read the lidar's datas on the Mission Planner interface but the drone doesn't to change its altitude while passing over an obstacle.

Being fairly new to DIYdrones community I though I would ask you some advices in order to have other opinions while I'll be looking into it.

EricXiao1990, 2016-02-26:
Hi, Benbenbeni.
       You are right, Lidar don't work now. Maybe this function will be added in the future.

Manubarret, 2016-03-06:
Hi,
I do agree that it would be a great feature! And hopefully not too complicated to adapt from loiter mode.

jward7, 2016-04-09:
Lidar (sonarrange) in auto mode would be a fantastic feature. It would greatly increase terrain following accuracy. We use a lightware SF10/c. Works great in loiter and alt hold, but does not work in auto.

magicrub, 2016-04-09:
This has been discussed several times and is on the copter roadmap.

rmackay9, 2016-09-08:
This is include in Copter-3.4 and there's a wiki page describing it here so I'll close this issue if that's ok.
http://ardupilot.org/copter/docs/terrain-following.html

closed 2016-09-08 by rmackay9