ArduPilot/ardupilot #3630: Increase stream rate limit beyond 50Hz

magicrub, 2016-02-17:
Specificilly, get the IMU data out as fast as possible

allanm84, 2016-06-19:
Has anyone tried increasing the gcs_data_stream_send task beyond 50hz?  Is that not sufficient to get state information out?  I was hoping to do exactly that but I figured it must be limited to 50hz for a reason...  If it cannot be increased, perhaps a separate task is created that spits out IMU data over the mavlink port at a faster rate?

magicrub, 2016-06-19:
It is possible to raise this up to the ekf rate with only a small change,
just need to be careful about is implications.

For plane and rover that's 50hz. For quadplane that's 300hz (flexible) and
Copter it's 400hz.

allanm84, 2016-06-19:
What are the implications?  It seems like running at 400hz is a bad idea as one could pretty quickly saturate the serial buffer(s) depending on how many streams are being requested by GCSs and what the baudrate is.  Is there a sense of what would be a safe rate to increase the gcs_data_stream_send task to?  Or better yet, whats the best way to determine if an increased rate is "safe"?

magicrub, 2016-06-20:
There is a self regulation of steam rates. It will automatically scale back
the rates if  the tx buffers fill up.

But yeah, setting it to 400 would be bad on certain streams but it is
recoverable so it's probably ok.

rmackay9, 2016-06-20:
I think we will probably need this as we start doing SLAM / Non-GPS navigation with companion computers where the external EKF needs the accel + gyro data at a high rate.  I think consumption of incoming messages should also be increased to 400hz.

allanm84, 2016-06-20:
@rmackay9 You're saying we should up the gcs_data_stream_send to 400hz?  It looks like gcs_check_input is already running at 400hz.

allanm84, 2016-06-20:
After examining the code a bit, it looks like GCS_Common.cpp expects the stream_trigger() to be called at 50hz: https://github.com/planckaero/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L666.

WickedShell, 2019-09-03:
This has been superseded by the per message interval stuff, which is superior as you can get just the IMU messages you want, and you can run up to 80% of the main loop rate on these.

closed 2019-09-03 by WickedShell