ArduPilot/ardupilot #3641: Pixracer: No PWM7 for tricopter yaw servo

MR444, 2016-02-18:
Hello at github!
Just finished my rebuild tricopter with pixracer flight controller but cannot fly!

APM Wiki says that yaw servo has to be connected to PWM 7.
http://copter.ardupilot.com/wiki/tricopter/
But Pixracer only has 6 pwm outputs.

How to connect yaw servo?
I can not find any parameters on Missionplanner, APM Planner or QGroundControl.

rmackay9, 2016-02-19:
Yes, that is trouble indeed.  In Copter-3.4 we have the flexibility of moving the motor mapping around.  If you're comfortable running master (aka "latest", aka "trunk") then you can load that up from the mission planner's Firmware Install screen after first pressing Ctrl-Q (or maybe it's Alt-Q).
Then you should be able to set the RC6_FUNCTION to 39.
I haven't tested this but I think it should work.
Note that the pixracer is a bit new and there are a couple of issue we're working through.  In particular a "Bad Gyro Health" message.  It's a bit of a false positive on a check we do on the IMU so it's safe to fly but it's annoying and may stop you from being able to arm.  Discussion is here: http://diydrones.com/forum/topics/pixracer-with-ardupilot

rmackay9, 2016-02-19:
By the way, "master" is use-at-your-own-risk because it hasn't gone through all the testing that an official release has.  Also it changes nearly every day as new fixes, enhancements go in, sometimes new bugs are (hopefully only temporarily) introduced.  This means it's not something that we support.  I.e. we can't keep everyone informed of the changes that are going in.

MR444, 2016-02-19:
Thanks, I hav installed  3_4-dev Tri via Missionplanner.
RC6_FUNCTION must be set to 39? Is it the correct value? So only description of function 39 is still missing? 

Maybe you could also add RC_FUNCTION for channel 3 because it is not in use with tricopters. Otherwise there would be only 1 free PWM output for tricopters with Pixracer (CH5).
[image]

MR444, 2016-03-20:
Hi, is there something new about using RC_FUNCTION for other channels than RC_6? 

I would like to replace my big pixhawk by a pixracer at my tricopter but I would need all 6 PWM outputs, so RC_FUNCTION for channal 3.
Thanks!

Huege, 2016-03-24:
Hello,  
with the latest master it seems not to work, no output on servo7_raw  and no effect with RC6 set to 39. I was successful in compiling Arducopter tri with adjusted AP_MotorsTri.h to RC6 one week ago . I have the yaw servo output on pin 6 now. Unfortunately facing other issues which I have to figure out.
I would prefer to download a compiled version from here until AC 3.4 reaches beta status.

Thanks in advance!

MR444, 2016-05-08:
Can anybody tell me if it is working at the moment with 3.4 dev or another version? I do not want to exchange flight controller again if it is not working.

Huege: Which compiled version do you suggest?

OXINARF, 2016-05-08:
I believe this works on master. @tridge I think you have done some work on this, can you confirm it is working?

Huege, 2016-05-09:
It's definitely working! I tried it one weeks ago.
Cheers

tridge, 2016-05-09:
yes, it works in master
there is still no RCn_FUNCTION for channels 1 to 4 though
i'll close this as it is definitely working for tricopter with tail on chan6

closed 2016-05-09 by tridge

tridge, 2016-05-09:
I've also added the parameter docs for these RCn_FUNCTION values, so they should appear in the drop down menu in MissionPlanner soon (ie. RC6_FUNCTION=Motor7)

MR444, 2016-06-16:
Hi AnadBiradar,
RC6_Function only works 3.4. I did not compile anything but it works with 3.4 (I still have dev)

AnandBiradar, 2016-06-22:
Thanks @MR444, I ran Copter 3.4 (my compiled version) on my Pixracer powered Tri but Motor 1 (channel 1) didn't move after arming. I saw the same behavior for Copter-3.3-Pixracer.

Any possibility you experience something like this ? Also if it's not too much to ask would you be able to share your version copter-3.4 ?