ArduPilot/ardupilot #3642: Plane: reverse thrust RC3_TRIM=1200 broken

magicrub, 2016-02-18:
@iskess is doing reverse-thrust testing and reported that a throttle trim of 1200 did not work but 1500 did. I tested and ti worked OK but I think I'm testing it differently than his setup. @iskess please report the configuration and what it's doing wrong.

iskess, 2016-02-19:
I've changed my ESC, RC Tx, and APM parameters to accommodate a 1500 neutral so I can't do further testing without resetting everything back. I reported on PR #3567 the following:

"With RC3_TRIM set to 1200 I'm seeing the Throttle as 57% at neutral. As I add forward throttle that number decreases until PWM 1500 where it shows 0% Throttle, then as I continue to advance the throttle it increases up to 100%. Reverse throttle shows 57% then increases upward to 100% at full reverse which is a range of 1200-1000 PWM."

So to summarize:
PWM 1200 -> 1500  =  Throttle 57%  -> 0%
PWM 1500 -> 2000  =  Throttle 0%  -> 100%
PWM 1200 -> 1000  =  Throttle 57% -> 100% (reverse)

If further information is needed, I will reconfigure everything, its the least I could do for all the work you are putting into this Tom. However if you can't duplicate it, perhaps we should wait and see if someone else can. Maybe I missed something.

magicrub, 2016-02-19:
The reported throttle to the gcs may be the issue instead of the actual
throttle. I'll just keep an eye out on this.

iskess, 2016-02-19:
Yes that could be it. The CH3 on the tuning graph reported correctly. I couldn't find a Throttle% on the tuning graph or on the Status page. I was using the HUD User_Items as you said.

OXINARF, 2017-06-04:
@iskess Is this still valid with latest Plane release?

iskess, 2017-06-04:
I'm not set up to test it right now, but since it's been open for over a year without any other complaints I suppose it's safe to close.

magicrub, 2017-06-04:
Thanks Iam. I'll close it but if you find it still broken then please reopen.

closed 2017-06-04 by magicrub