ArduPilot/ardupilot #3644: New Flight Mode - Dron goes ahead of us a certain distanc

bitusbates, 2016-02-19:
Hello,
I'm trying to do a new flight mode with an APM 2.6, in order to facilitate passage of ambulances and other emergency vehicles. That's why I want to create a flight mode that do that my drone stays always a certain distance from me or a vehicle (for example 100 meters). I will use a telemetry module with my mobile, and a U-Blox 6M NEO GPS in my drone. The idea is to set some waypoints with the Tower App, and then through this "New Flight Mode", do that the drone stays always these 100 meters (radius) from me or vehicle. See please the image below to undestand the situation easier ...
Could someone help me to create this new mode?
Thank you so much,
David

marcmerlin, 2016-02-22:
Probably not what you want to hear, but APM 2.6 is totally full, there is no room to add anything to it without removing other features.
This means if you're stuck with that, you'll have to write the feature yourself and maintain your own tree where you'll remove other features you can live without to make room for your new features.
Otherwise you'll want to have at least a pixhawk.

bitusbates, 2016-02-22:
Hi Marc,
Thanks for your answer. Do you know how could I build this function? Which file should I modify or create etc?  I think the idea is quite similar to the follow-me mode, but I need that the drone goes ahead of me a certain distance, always following a route given previously by me.
Thanks again,
David

magicrub, 2016-02-22:
Sounds like normal waypoint navigation but with a follow-me leash

bitusbates, 2016-02-22:
Yes.
But I need the drone goes ahead of me instead behind of me. I don't know if
I can write a leash with negative numbers, in order it goes ahead of me...

Thanks,
David

magicrub, 2016-02-22:
I don't see anyone implementing this just for fun, or just because you
asked nicely. Who knows, maybe you'll get lucky (eventually). However, this
is a great example of where hiring an ArduPilot developer to do a very
specific feature request may be worthwhile. Allows you to work with the
developer to implement it exactly how you'd like it to be done.

bitusbates, 2016-02-22:
Hi Tom,

I'm doing that for as final degree project, that's why I would like to
learn how implement new functions to my APM, and this specifically. If
someone could help me, I'll be very grateful for the help.

Thanks,

David

mikerob, 2016-02-22:
https://3dr.com/kb/follow-instructions/ contains references to a "lead" mode where the copter flies in front of the operator - isn't that what you're looking for?

bitusbates, 2016-02-22:
Yes, this is more or less the idea what I'm looking for. However I need it
goes in front of the operator but following a route given by me
(waypoints). I mean it'a not the same if it goes 100 meters in front of me
in an arbitrary direction, instead of 100 meters in front of me, following
a route. If I could achieve this, then it would be perfect!
Hope someone could help me :)
Thank you,
David

hiro2233, 2016-09-08:
@bitusbates sound very interesting your plan mode with gps waypoints, i'm sure that you will get help if you post in our new discuss forum http://discuss.ardupilot.org/c/arducopter/copter32

Thanks for your participation.
Good luck!

@OXINARF please close this.
Thanks.

closed 2016-09-09 by magicrub

jberaud, 2016-09-09:
@hiro2233 Dear Hiro. I don't think it is appropriate to ask @OXINARF to close pull requests. I guess if he has reasons to do so, he will.

hiro2233, 2016-09-09:
@jberaud Ok dear Julien, sorry, i thought that was where he could close it. but then, who can i call?

hiro2233, 2016-09-09:
@magicrub thank you for the favor, indeed.

jberaud, 2016-09-09:
my mistake, pull requests and/or issues.
@hiro2233 I would say that you can write that to your opinion this issue/pull request can be closed.
I guess that the project maintainers could also give you the permissions to close issues/pull requests if they find it appropriate.

hiro2233, 2016-09-09:
@jberaud ahh ok, i understand. I will write on that manner. Thank you for the advice Julien.

rmackay9, 2016-09-09:
Here's a related issue involving adding the follow-me mavlink message which an on-board follow me flight mode could use (as opposed to the GCS based follow-me in Tower).
https://github.com/ArduPilot/ardupilot/issues/3929