ArduPilot/ardupilot #3645: Redundant RC_CHANNELS telemetry data.

AndKe, 2016-02-19:
observed on current arducopter release:
[image]

marcmerlin, 2016-02-22:
Are you sure this is not a UI bug in mission planner or whatever GCS you're using?

AndKe, 2016-02-22:
yes, that was MAVlink inspector widget of APM planner (so I see update rates)  , but here is more visual dump with time..
```
2016-02-16 10:43:17.11: RC_CHANNELS {time_boot_ms : 986955, chancount : 18, chan1_raw : 1496, chan2_raw : 1488, chan3_raw : 982, chan4_raw : 1495, chan5_raw : 982, chan6_raw : 982, chan7_raw : 982, chan8_raw : 1494, chan9_raw : 874, chan10_raw : 874, chan11_raw : 874, chan12_raw : 874, chan13_raw : 874, chan14_raw : 874, chan15_raw : 874, chan16_raw : 874, chan17_raw : 998, chan18_raw : 998, rssi : 0}
2016-02-16 10:43:17.58: RC_CHANNELS_RAW {time_boot_ms : 987455, port : 0, chan1_raw : 1496, chan2_raw : 1490, chan3_raw : 982, chan4_raw : 1497, chan5_raw : 982, chan6_raw : 982, chan7_raw : 982, chan8_raw : 1494, rssi : 0}
2016-02-16 10:43:17.61: RC_CHANNELS {time_boot_ms : 987455, chancount : 18, chan1_raw : 1496, chan2_raw : 1490, chan3_raw : 982, chan4_raw : 1497, chan5_raw : 982, chan6_raw : 982, chan7_raw : 982, chan8_raw : 1494, chan9_raw : 874, chan10_raw : 874, chan11_raw : 874, chan12_raw : 874, chan13_raw : 874, chan14_raw : 874, chan15_raw : 874, chan16_raw : 874, chan17_raw : 998, chan18_raw : 998, rssi : 0}
2016-02-16 10:43:18.08: RC_CHANNELS_RAW {time_boot_ms : 987955, port : 0, chan1_raw : 1496, chan2_raw : 1490, chan3_raw : 982, chan4_raw : 1496, chan5_raw : 982, chan6_raw : 982, chan7_raw : 982, chan8_raw : 1494, rssi : 0}
```

marcmerlin, 2016-02-22:
Thanks for confirming. I'm only helping with triage, but indeed it looks like this should be fixed.
If I had to guess, this was left for backwards compatibility with old GCS but maybe this can be removed now?

marcmerlin, 2016-03-29:
@WickedShell @magicrub can you chime in on this one, and potentially assign?

IamPete1, 2021-07-25:
Seems to be a comparability thing for some old OSD's see:
https://github.com/ArduPilot/ardupilot/blob/299b3ef2009f1e5a43b394e826dce7d445d30fc6/libraries/GCS_MAVLink/GCS_Common.cpp#L1662

We only send on MAVLink1, MAVLink2 is the default now.

IamPete1, 2021-08-08:
I'm going to close this, were sending extra data by design and this is not a issue with MAVLink2 so there is little benefit to removing the extra message.

closed 2021-08-08 by IamPete1