ArduPilot/ardupilot #3655: APM Plane, Pitch / Throttle spikes

SlipstreamFPV, 2016-02-21:
I've just noticed some strange spikes which build up on my pixhawk APM plane 3.5 logs. When I fly in cruise mode the spikes increase which causes pitch servo oscillations and throttle pulsing. I'm not sure if this is a bug or just a problem with my setup.  I upgraded from a clean flash of APM Plane 3.4 which was working well as you can see below: 

APM plane 3.4:
[image]

APM plane 3.5:
[image]

APM plane 3.5 showing effect on throttle output:
[image]

APM plane 3.5 showing effect on pitch servo output:
[image]

Obviously this leads to the motors and escs heating up and the pitch servo getting a workout which isn't ideal and could lead to a failure / crash. I hope it's not a bug but it seems strange when It wasn't happening before. Vibration levels average around 1.6 which is nothing.

I can provide a link to a log of the APM Plane 3.5 flight if requested. My parameter file:
[MTD.txt](https://github.com/diydrones/ardupilot/files/139444/MTD.txt)

SlipstreamFPV, 2016-02-22:
Anyone have an opinion whether this might be a bug? It seems like one of the parameters is getting out of control and escalating.

marcmerlin, 2016-02-22:
Stupid question maybe, but to rule out that nothing changed in the hardware/build, did you try going back to 3.4 and confirmed that this goes away?

SlipstreamFPV, 2016-02-27:
Nothing changed on the build. This problem is doing my head in. I've tried changing things back and forward and the issue persists. It would be awesome if someone could look at my log and at least confirm this isn't a bug. If it is it a bug it could easily overheat the servos / Esc and bring a plane down.

SlipstreamFPV, 2016-02-27:
https://www.dropbox.com/s/abj7urzimktzdto/2.BIN?dl=0

marcmerlin, 2016-02-27:
just to set the expectations, my understanding is that there are only 2 or 3 devs who can look and make sense out of this (at most) and they are quire busy/behind as is, so it may take a while, or not happen. I may be wrong, but I wanted to give you a possible head up.

magicrub, 2016-02-27:
Try again with your tecs-spdwght set to zero. I was having trouble with
this to and decoupling then helped. Also try lowering you arspd-fbw-min
which may be inducing it too.

SlipstreamFPV, 2016-02-27:
Thanks a lot for taking the time to write back Tom. I'll give it a go. The min cruise speed is very close to the actual cruise speed so that might be the culprit right there. I did pull the max cruise speed down quite a bit when I upgraded too so maybe that effected the ratio on the lower end too. Hopefully it'll get my Pixhawk twin dream back in the air so I can keep testing the new features and releases that you guys work so hard on.

SlipstreamFPV, 2016-02-28:
Thanks for your suggestions.. It appears as though the problem is the same or even a little bit worse with some adjustments to cruise speed and TECS_SPDWEIGHT. The log is here:

https://www.dropbox.com/s/sfbm41ro714iq8d/4.BIN?dl=0

If you have the time to take a look it would be great. I'd rather not completely decouple the TECS_SPDWEIGHT as there are obviously benefits to leaving it enabled for better plane control so I left it on 0.5. I imagine the problem is occurring higher up in the control loop anyway. I'd much rather find the source of the issue :)

[image]

magicrub, 2016-02-28:
What type of airplane are you flying?

SlipstreamFPV, 2016-02-28:
"My twin dream" It's a twin tractor prop. 1.8M wingspan.

SlipstreamFPV, 2016-03-01:
Seems others have similar graphs resulting from GPS lock. I don't know if having 2 GPS modules reporting slightly different things might have a similar effect but I'm convinced something is feeding wrong data into the control loop since it's asking for these odd corrections.

SlipstreamFPV, 2016-03-01:
http://diydrones.com/forum/topics/fbwa-servo-jitter-when-gps-has-no-fix

magicrub, 2016-03-01:
Jitter with no gps fix is totally different than a feedback system a little
out of whack

SlipstreamFPV, 2016-03-01:
Thanks for the clarification. I just have no idea where to start looking to solve this problem. I have looked through all the logged data traces and these ones stand out as having bad osscilations while in Cruise mode:

[image]

[image]

[image]

but I couldn't find documentation on what these graphs are and which parts of the tuning loop take part in each.

magicrub, 2016-03-01:
H is height, SP is airspeed, -dem means demand, d in front means derivative
(change from last sample).

SlipstreamFPV, 2016-03-02:
So the first graph indicates that the autopilot is requesting way too many speed changes then? What would cause it to do that?

closed 2016-03-06 by SlipstreamFPV

magicrub, 2016-03-06:
Any conclusions or insights?

SlipstreamFPV, 2016-03-06:
I reset everything again and did all the calibrations one more time and still had some issues although not as bad. I eventually noticed that the hitec servo (HS65MG) I had on the ele was jittering so I switched it out for the usual turnigy ones I use (TGY-50090M). Don't know how I didn't notice it sooner but it seems to have had a big effect on tuning loop. The throttle and pitch loops may have been playing catch up on the very small random pitch angles commanded by the jittering servo. Please have a look at the new log and see what you think:

https://www.dropbox.com/s/bxbviuh7abb3u5t/9.BIN?dl=0

The compasses appear slightly off in the logs but don't throw any EKF errors through mission planner, is that normal? Thanks for looking into the issue, sorry I had us chasing nothing in the end embarrassingly :(

wazoo22, 2018-11-13:
Interesting problem. I've had the same problem using a Pixhawk 1 on Plane 3.5 and on a different controller (No names, but it rhymes with Beagle Fleas) and I've heard several aircraft in cruise or similar mode with cycling throttle- sometimes fast, sometimes slower. There's more to this, I think.
When you say "jitter", roughly what amplitude, and how fast? Obviously strong enough to get your attention.