ArduPilot/ardupilot #3656: do_set_home command has no effect

oceangis, 2016-02-21:
i build a boat with arduRover with APMrover 2.5 firmware. I  test my boat i reset the home location through "set home here" in mission planner, when i execute RTL, the boat return to the GPS location first locked not the new home location.

salonijain12, 2016-03-11:
I am able to solve the issue #3656, could you please share the procedure to submit/close this bug as I am new to Ardupilot and not aware of the process.

magicrub, 2016-03-11:
The procedure is to make the fix locally, then rebase it onto master branch, then create a Pull-Request here https://github.com/diydrones/ardupilot/pulls and submit it for review. Once accepted into master this issue can be closed.

amilcarlucas, 2017-09-25:
Hasn't this been implemented last week ?

OXINARF, 2017-09-27:
This was for Rover, so it was fixed with #3747. Last week were just changes, namely supporting DO_SET_HOME command within COMMAND_INT.

closed 2017-09-27 by OXINARF