ArduPilot/ardupilot #3658: Assertion failed at file:armv7-m/up_hardfault.c line: 184 task: AHRS_Test

pyh0603, 2016-02-22:
I tried to run directory：\ardupilot\libraries\AP_AHRS\examples\AHRS_Test  Kalman filter test code, I can compile, a hardware error when I download to the flight control panel. I'm sure I'm using gcc-4.7 . the log is as follows:
AP_I2C on I2C bus 1 at 0x00 (bus: 100 KHz, max: 400 KHz)
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
[init] looking for microSD... 
[init] card mounted at /fs/microsd 
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[init] USB interface connected 
Running rc.APM 
Mounting binfs 
binfs mounted OK 
nsh: rm: unlink failed: No such file or directory
Detected FMUv2 board 
uorb started OK 
Trying PX4IO board 
PX4: param_find(SYS_RESTART_TYPE)
PX4: param_find(RC_RSSI_PWM_CHAN)
PX4: param_find(RC_RSSI_PWM_MAX)
PX4: param_find(RC_RSSI_PWM_MIN)
PX4: param_find(CBRK_IO_SAFETY)
px4io default PWM output device
PX4: param_find(CBRK_FLIGHTTERM)
PX4IO board OK 
PX4: param_find(RC_DSM_BIND)
PX4: param_find(BAT_V_SCALE_IO)
PX4: param_find(RC_FAILS_THR)
PX4: param_find(RC_FAILS_THR)
PX4: param_find(CBRK_FLIGHTTERM)
PX4: param_find(RC_RSSI_PWM_CHAN)
PX4: param_find(RC_RSSI_PWM_MAX)
PX4: param_find(RC_RSSI_PWM_MIN)
PX4: param_find(PWM_MAIN_REV1)
px4io: CRCs match
PX4: param_find(PWM_MAIN_REV2)
PX4IO CRC OK 
PX4: param_find(PWM_MAIN_REV3)
PX4: param_find(PWM_MAIN_REV4)
PX4: param_find(PWM_AUX_REV1)
PX4: param_find(PWM_MAIN_REV5)
PX4: param_find(PWM_AUX_REV2)
PX4: param_find(PWM_MAIN_REV6)
PX4: param_find(PWM_AUX_REV3)
PX4: param_find(PWM_MAIN_REV7)
PX4: param_find(PWM_AUX_REV4)
PX4: param_find(PWM_MAIN_REV8)
PX4: param_find(PWM_AUX_REV5)
PX4: param_find(TRIM_ROLL)
PX4: param_find(PWM_AUX_REV6)
PX4: param_find(TRIM_PITCH)
Set FMU mode_pwm4 
PX4: param_find(TRIM_YAW)
Starting APM sensors 
PX4: param_find(PWM_SBUS_MODE)
MS5611_SPI on SPI bus 4 at 2 (20000 KHz)
ms5611: no device on bus 4
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
ms5611: no device on bus 1
ms5611: no device on bus 2
ms5611 started OK 
adc init done
adc started OK 
Starting FMUv2 sensors 
HMC5883_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
Have external hmc5883 
hmc5883: no device on bus 1
No internal hmc5883 
mpu6000: driver start failed
mpu9250: driver start failed
No MPU6000 or MPU9250 external 
MPU6000 on SPI bus 1 at 4 (1000 KHz)
Found MPU6000 
L3GD20 on SPI bus 1 at 1 (11000 KHz)
l3gd20 started OK 
LSM303D on SPI bus 1 at 2 (11000 KHz)
lsm303d started OK 
ets_airspeed: no ETS airspeed sensor connected
meas_airspeed: no MS4525 airspeed sensor connected
meas_airspeed: no MS4525 airspeed sensor connected
ll40ls: driver start failed
ll40ls: driver start failed
trone: driver start failed
mb12xx: driver start failed
px4flow: scanning I2C buses for device..
started pwm_input driver 
started mtd driver OK 
reading /fs/mtd expecting 16384 bytes
readtest OK
mtd readtest OK 
oreoled: autoupdate enabled
oreoled on I2C bus 2 at 0x68 (bus: 100 KHz, max: 100 KHz)
oreoled started OK 
batt_smbus on I2C bus 2 at 0x0b (bus: 100 KHz, max: 100 KHz)
Found batt_smbus 
irlock: failed to initialize device, stopping driver
rwtest /fs/mtd testing 16384 bytes
rwtest OK
mtd rwtest OK 
Starting ArduPilot /dev/ttyACM0 /dev/ttyS1 /dev/ttyS2 
Starting AHRS_Test uartA=/dev/ttyACM0 uartC=/dev/ttyS1 uartD=/dev/ttyS2 uartE=/dev/ttyS6
initialised /dev/ttyACM0 OK 4096 1024
initialised /dev/ttyS3 OK 1024 512
initialised /dev/ttyS1 OK 1024 512
initialised /dev/ttyS2 OK 1024 512
initialised /dev/ttyS6 OK 1024 5ArduPilot started OK 
12
rc.APM finished 

NuttShell (NSH)
sample_rate: 100test sensor 1
nsh> [Ktest sensor 2
test sensor 3
test sensor 4
Assertion failed at file:armv7-m/up_hardfault.c line: 184 task: AHRS_Test
sp:     2000c798
IRQ stack:
  base: 2000c7dc
  size: 000002e8
2000c780: 08068288 000000b8 080681c2 2000c798 1000f8b8 080449bd 08068288 000000b8
2000c7a0: 0803e565 00000010 2000b10c 00000003 00000000 08044c4d 08044c39 0804ed7d
2000c7c0: 00000000 20005a50 20005afc 20005f2c 00000000 08044bbd 1000f794 00000000
sp:     1000f868
User stack:
  base: 1000f8b8
  size: 00001ffc
1000f860: 20000010 0800d733 0800d795 00000000 00000000 00000000 00a0793c 00000000
1000f880: 00000000 00000000 0800d795 00000000 00000000 00000000 00000000 00000000
1000f8a0: 00000000 00000000 00000000 0803f86b 00000000 00000000 ffffffff 0bc19624
R0: 1000d374 20008190 20008190 00000000 20005a50 20005afc 20005f2c 00000000
R8: 100049e0 1000d390 20005e78 00000000 1000f828 1000f868 0800d73b 00000000
xPSR: 60000000 BASEPRI: 00000000 CONTROL: 00000000
EXC_RETURN: ffffffe9

pyh0603, 2016-02-22:
When I  debug the code, fond just call the function "ins.init (100)", that is set the sensor sampling frequency, there will be similar problems. Run some test code directory will have the same problem: ardupilot \ libraries \ AP_InertialSensor \ examples \ INS_generic
Who can answer my questions, thank you very much！

Naterater, 2019-01-07:
Is this still a problem @pyh0603?

WickedShell, 2019-07-14:
Closing on account of massive changes (including the RTOS) since this was reported, and no response in 6 months.

closed 2019-07-14 by WickedShell