ArduPilot/ardupilot #3660: Downloadable JSon file for firmware listings

lvale, 2016-02-22:
As a follow up to this issue https://github.com/mavlink/qgroundcontrol/issues/2797

peterbarker, 2016-03-07:
Will be looking at creating a PR for this shortly:  https://github.com/peterbarker/ardupilot-1/tree/generate-manifest

MAVProxy module to grab manifests, manipulate them and allow downloading: https://github.com/peterbarker/MAVProxy/tree/generate-manifest

```{
    "firmware": [
        {
            "mav-type": "GROUND_ROVER", 
            "format": "hex", 
            "url": "https://firmware.diydrones.com/Rover/latest/apm1/APMrover2.hex", 
            "mav-firmware-version-type": "DEV", 
            "mav-autopilot": "ARDUPILOTMEGA", 
            "platform": "apm1", 
            "git-sha": "705d3d567abfe8d86cbc5f3120bc6b09b879476f", 
            "latest": 1
        }, 
```

Biggest issue I have right now is that it would be nice to have "mav-firmware-version" as a field in there, so someone can find e.g. Copter 3.3.2; I expect to take the information from the FIRMWARE_VERSION define and make it available in the manifest.

lvale, 2016-03-07:
@peterbarker ardupilot.org not diydrones.com :)

DonLakeFlyer, 2016-03-07:
So this doesn't look like what was discussed in the Issue? How does it relate to that?

peterbarker, 2016-03-10:
@DonLakeFlyer The generate-manifest branch generates a JSON string, a snippet of which I pasted in.  The mavproxy module is a rough example of how to use it.

DonLakeFlyer, 2016-03-10:
So, some of the information needed is in there. But it's missing stuff.

peterbarker, 2016-03-10:
@DonLakeFlyer Yes, I've mentioned some of the information there - the FIRMWARE_VERSION information.  Can you think of what else might be required and/or useful?

closed 2016-04-20 by lvale