ArduPilot/ardupilot #3661: Copter: give altitude priority over yaw.

AndKe, 2016-02-23:
This is one recent crash due to loss of thrust on a octo that theoretically could fly with constant yaw:
http://ardupilot.com/forum/viewtopic.php?f=100&t=15070

Autopilot is perfectly able to fly with constant yaw, the problem seems to be that yaw is kept at the cost of needed thrust.

This would be a great safety feature, allowing those who can't fly with constant yaw , or being too far away to do that, an good ending.

lthall, 2018-01-11:
This can only really be done after a motor failure is detected. Without motor failure the yaw output should not compromise throttle in a significant way.

IamPete1, 2021-08-30:
@lthall I think you have made significant improvements to the yaw failure since this report. 

https://github.com/ArduPilot/ardupilot/pull/17209
https://github.com/ArduPilot/ardupilot/pull/12700
https://github.com/ArduPilot/ardupilot/pull/12696

So maybe we can close this?

rmackay9, 2021-08-31:
Yes, this is done so let's close it.

closed 2021-08-31 by rmackay9