ArduPilot/ardupilot #3675: MAV_CMD_CONDITION_DELAY enhancement

robustini, 2016-02-26:
I need to perform automatic events at set times.
Chatting with Randy was right to me to be simple, and it would be interesting to implement it.
Using "MAV_CMD CONDITION DELAY" where instead of declaring the seconds of pause just enter a UTC time in 24h format (es. 13:00:15), detected by GPS so absolutely reliable.
I believe that this feature will be very interesting, especially in the swarming and autonomous flights.

magicrub, 2016-02-26:
+1 Love the idea!

robustini, 2016-02-26:
An alternative, to avoid using too many events, would be to be able to set a fixed time for each type of event, but perhaps at programming level would be more complicated.
I hope Randy and Michael implement as soon as possible this thing.

DonLakeFlyer, 2016-02-26:
My only comment would be to implement it as a new command, not as a one-off of MAV_CMD_CONDITION_DELAY

robustini, 2016-02-26:
MAV_CMD_CONDITION_TIME...?

rmackay9, 2016-04-06:
After our discussion on the dev call I think we decided to create a new navigation command (instead of a "do" or "conditional" command).  So a good name might be MAV_CMD_NAV_DELAY.

Here's another related issue: https://github.com/ArduPilot/ardupilot/issues/3675

robustini, 2016-04-08:
Really nice, i wait the integration (also inside the Planner), thanks Randy!

rmackay9, 2016-08-29:
Closing this issue because this was implemented with the NAV_DELAY command.  I've added the need to add documentation in this other issue.  https://github.com/ArduPilot/ardupilot/issues/4674

closed 2016-08-29 by rmackay9