ArduPilot/ardupilot #3683: How to make a new flight mode appears on MAVProxy?

nocternal, 2016-02-27:
I'm playing with Skywaker X8 plane, and want to create a new flight mode to do some stunts XD,and I want to test the code first by SITL with MAVproxy, but even the first step: let the mode appear on the MAVproxy's available modes list failed.Here are the things I have done:

1.  Define for the new flight mode in defines.h [https://github.com/nocternal/ardupilot/blob/bug/ArduPlane/defines.h](url)
`QHOVER = 18, 
QLOITER   = 19, 
JULAND     = 20, //in enum flight  FlightMode ,my new flight mode called JULAND,and give value 20`

2. Add mode name in ArduPlane/APM_Config.h.reference   6edb9fa  
`   //  MANUAL          | Full manual control via the hardware multiplexer.  
    //                           |  
  +//  JULAND           | very original ju style.  `

3. Add new flight mode's option in ArduPlane.cpp   , now I just add simple codes to verify it works in the MAVproxy. and also try my flightmode with MANUAL's option, but not work in MAVproxy either
 f1b709d  
`+case JULAND: 
+ channel_roll->servo_out = 2000;  
+ channel_pitch->servo_out = 1100;  
+ steering_control.steering = steering_control.rudder = channel_rudder->pwm_to_angle();  
+ channel_throttle->servo_out = 100;  
break;   `

4. Add mode case in ArduPlane/GCS_Mavlink.cpp  6edb9fa 
case JULAND:  //do nothing as other modes

5.Add mode in ArduPlane/Parameters.cpp ,make it  as a choice in RC switch 1~6 672d090 
`// @Param: FLTMODE2  
// @DisplayName: FlightMode2  
 // @Description: Flight mode for switch position 2 (1231 to 1360)  
// @Values: 
0:Manual,1:CIRCLE,2:STABILIZE,3:TRAINING,4:ACRO,5:FBWA,6:FBWB,7:CRUISE,8:AUTOTUNE,10:Auto,11:RTL,12:Loiter,15:Guided,20:JULAND `

6. Add mode case in ArduPlane/system.cpp d092561  

After done these, when do SITL in MAVProxy, to input "mode" on console ,get this:
'Available modes:',['RTL','TRAINING','LAND'.........but my new flight mode not appear.
So I think maybe Mavproxy's code also need some change,so I download mavproxy's source code ,
and find seems that I need also add code in ardupilot/modules.mavlink/pymavlink/mavutil.py:
`mode_mapping_apm = {
    0 : 'MANUAL',
    1 : 'CIRCLE',
    2 : 'STABILIZE',
    3 : 'TRAINING',
.......
    18 : 'QHOVER',
    19 : 'QLOITER',
    20 : 'JULAND'
    }`
but MAVPROXY still not has my new mode JULAND
 
so I decided to ignore this and set flightmode 5 as 20 (20 presents my mode ),as a myparameter.parm. After take off successful with AUTO, I input "switch 5" in console.
and a error comes out : 
"Verify nav. Invalid or no current nav cmd"
this error is part of code in ArduPlane/commands_logic.cpp 
Plane::verify_command(const AP_Mission::Mission_Command& cmd)'s part of code......

I'm confused what further steps needs to be done= =
the main question are these three:
1.Am I add a new flight mode right in plane's code ?For there is only copter's example
2.How to make a new flight mode available in Mavproxy?
3.Why "Verify nav. Invalid or no current nav cmd" happens?

rmackay9, 2016-08-10:
I think maybe you need to do an "make install"?  I'm guessing..

gmorph, 2016-08-10:
From the pymavlink folder run
python setup.py build install --user
If for some reason you have installed it system wide ( I would advise against this) you will need to drop the --user bit.
Hope this helps.

closed 2017-08-09 by nocternal

matteoscanavino, 2018-11-06:
@nocternal  Hello, were you able to solve the problem? I'm facing with the same problme with ArduCopter. I've built my flight mode but I cannot see it in SITL mavlink...

auturgy, 2018-11-06:
@matteoscanavino would be better to ask at discuss.ardupilot.org
Perhaps read here: http://ardupilot.org/dev/docs/apmcopter-adding-a-new-flight-mode.html if you haven’t seen it before

rmackay9, 2018-11-06:
By the way, you don't actually have to be able to see the mode to change to it in SITL.  if you type "mode XX" where "XX" is the number of the mode you can switch into it and test it.

matteoscanavino, 2018-11-06:
@rmackay9 Thanks a lot, this solved

peterbarker, 2018-11-06:
On Mon, 5 Nov 2018, matteoscanavino wrote:

> @nocternal Hello, were you able to solve the problem? I'm facing with the same problme with ArduCopter. I've built my flight mode but I cannot see it in
> SITL mavlink...

Did you see step 7 here: 
http://ardupilot.org/dev/docs/apmcopter-adding-a-new-flight-mode.html ?

matteoscanavino, 2018-11-06:
@peterbarker Yes I read it. But where is located the file mavlink/ardupilotmega.xml? I cannot find inside the root folder ardupilot neither inside ArduCopter... do you know where is it?

auturgy, 2018-11-06:
https://github.com/ArduPilot/mavlink/tree/master/message_definitions/v1.0

Regards,

James

matteoscanavino, 2018-11-06:
@auturgy Thank-you!

schoningerJim, 2019-01-29:
> @rmackay9 Thanks a lot, this solved

Dear @matteoscanavino ,
Which solution worked for you?
did u modify something in mavlink sourcecode? or u just compile ardupilot with ur new mode and use mode XX to change into this mode?
can it work with SITL?
Thank you

yongqin12, 2019-07-25:
i think @grorph is right,i use python setup.py build install .and then the problem is solved

lifeof3point1416, 2019-11-06:
Hi, I also tried to add a flightmode to ArduCopter (version 3.6.11)
I did the 7 steps of ###, including adding it to ardupilotmega.xml
after I couldn't switch to my new flightmode in mavproxy, using sitl, i also tried @gmorph s
`python setup.py build install --user`, but it still didn't work. 
Next I edited `ArduCopter/config.h`, which is **strongly advised against** in the file's top part.
I added the define for my flightmode

    #ifndef MODE_MEASUREMENT_ENABLED
    # define MODE_MEASUREMENT_ENABLED ENABLED
    #endif

after the analogue define of the throw flightmode. (`MEASUREMENT` being my newly added flightmode with number 27). Now it in sitl when typing 

    mode 27

, as suggested by @rmackay9, it seems to switch the flightmode successfully. But the flightmode is still not detected by name. So

    mode measurement

fails - however I can live with that ;) 

Is there any way that the MODE_<new flight mode>_ENABLED gets added to config.h automatically, or is this the way to add it?

Regards, Peter

0b10010010, 2020-05-21:
Hello,

If anyone is still wondering about adding the mode you created to the list of mode so you don't have to always type in mode xx, where xx is the mode number you added, editing mavutil.py in ardupilot/modules/mavlink/pymavlink/ then running `python setup.py build install --user` as gmorph said should do the trick. I had to close out of all my terminals then running the SITL again made my new mode's name visible when you type `mode` in MAVProxy console.

yongqin12, 2020-07-23:
Oh!thank you for your help! acqin 在校大学生 邮箱：bao2bei@sina.cn 签名由 网易邮箱大师 定制 On 05/22/2020 03:16, Alex Kim wrote: Hello, If anyone is still wondering about adding the mode you created to the list of mode so you don't have to always type in mode xx, where xx is the mode number you added, editing mavutil.py in ardupilot/modules/mavlink/pymavlink/ then running python setup.py build install --user as gmorph said should do the trick. I had to close out of all my terminals then running the SITL again made my new mode's name visible when you type mode in MAVProxy console.

yongqin12, 2020-09-11:
Hello, how to add an optional servo parameter to the APM firmware and display it on the ground station。 Now the parameter added can realize the function, but it can't be displayed on the ground station bao2bei 邮箱：bao2bei@sina.cn 签名由 网易邮箱大师 定制