ArduPilot/ardupilot #3685: Support CRCs for parameter sets to speed parameter download to ground station

DonLakeFlyer, 2016-02-28:
QGC+PX4 stack has been running a new protocol which works on top of/using the existing Parameter Request mavlink protocol to exchange CRC’s for parameter sets. If the CRC matches there is no need to download the full param list. It’s really nice to connect to a Vehicle and have the ground station ready to go almost instantaneously. This has been running in the QGC+PX4 combination for a couple months now. Last tweak to protocols explained here: https://github.com/mavlink/qgroundcontrol/pull/2821

It has other interesting implications which is that you now can have vehicle information while the vehicle is disconnected. For example you can plan a mission while disconnected from the vehicle but still have access to parameters so you can calc things like mission time and battery usage. You can also know whether the mission is for fixed wing or multi-rotor which changes some of the command capability.

Would be great if ArduPilot could support this as well.

meee1, 2016-02-28:
this isnt the first time this has comm up, and it is not currently archivable. because there are params what change on every boot. such as baro temp, and baro ground pressure.

DonLakeFlyer, 2016-02-28:
Two possibilities:
1) Support a volatile meta data flag
2) Move those out of parameters and into mavlink

meee1, 2016-02-28:
the reason they are parameters is that the ground truth can be change as needed.

WickedShell, 2016-02-29:
Does anyone safely adjust those at the moment? It seems to me like most of
that could be moved into mavlink controlled settings (particularly thinking
of the baro ones like that).

I'm under the impression that simply adjusting some of the values will
cause some large step inputs that could make the EKF unhappy, meaning that
it shouldn't be done in general regardless?

auturgy, 2021-02-10:
Fast param download is implemented via https://github.com/ArduPilot/ardupilot/pull/13911 although not yet supported by QGC

peterbarker, 2021-07-30:
Considering the speed and simplicity that the parameter-fetch-via-mavftp code gives, I doubt we would merge an implementation of this checksuming protocol - based purely on flash cost.

The ftp case code has some other significant advantages - working in the general case of some-parameters-changed, for example.

Taking to a DevCall to make a decision so expectations can be set.

DonLakeFlyer, 2021-08-01:
> Considering the speed and simplicity that the parameter-fetch-via-mavftp code gives, I doubt we would merge an implementation of this checksuming protocol - based purely on flash cost.

Speed, yes. Part of the mavlink spec, no. Hence QGC will likely never support.

tridge, 2021-08-02:
we will submit the (much better) ftp protocol to mavlink

closed 2021-08-02 by tridge