ArduPilot/ardupilot #3690: ArduPlane 3.5 firmware doesn't work with FlightGear SITL

jinchengde, 2016-02-29:
Hi
  I'm using ArduPlane 3.5 firmware for SITL with FlightGear, but the board is still in Initliseing States after we start the Sim Link. change the version to 3.4 is OK.
  Does anyone see that before?

magicrub, 2016-02-29:
I'll let @tridge chime in on this before I close but I think HITL (Hardware-in-the-loop) is generally broken right now. The SITL tools have gotten good enough that we have no longer needed it and so it has not been maintained properly.

jinchengde, 2016-03-01:
@magicrub @tridge 
I means the HITL with FlightGear, not SITL , sorry for mistake.
I think the HITL is more important for us because we can test it with Dronekit and observer the servo action. hopefully to hold this function in further.

magicrub, 2016-03-01:
SITL will work with dronekit and you can plot and log the servo output.

closed 2017-12-25 by jinchengde