ArduPilot/ardupilot #3691: how to modify the default ch3 pwm out?

oceangis, 2016-02-29:
I am building  a boat with ardurover 2.5 and bidirectional ESC. When i power on Pixhawk first , motor will run at high speed before powering RC. i have check the log file and found there is low pwm output with 1000 and think these is the problem. where i can modify the 1000 to 1500. 

thanks.

magicrub, 2016-02-29:
Set your rc3-trim to 1500.

closed 2016-02-29 by magicrub

WickedShell, 2016-02-29:
There was a bug where pixhawk would momentarily send 1000 on initial start
up, but that was fixed on master about a month ago.

oceangis, 2016-03-01:
Hi WickedShell,
    I think this is a bug and want to modify the firmware of Ardurover 2.5, can you tell me how to modify the code or the thread about this bug

WickedShell, 2016-03-01:
It's a very brief single cycle of 1000 pwm that gets out. The exact commit
for it is
https://github.com/diydrones/PX4Firmware/commit/22b6f5752ef0cb989a383f9b05bfce9509b35daa
its for the PX4Firmware layer. Just compiling off of master will fix it.
Talking to Grant will probably lend more help then me with rover :)

oceangis, 2016-03-01:
Hi,WickedShell,thank you very much!

oceangis, 2016-03-01:
change the default value from 1000 to 0 ,the output value is also 1000,