ArduPilot/ardupilot #3697: LightWare SF10/x via I2C is still not working in Copter-.3.3.3

abencomo, 2016-03-01:
I'm running APM:Copter V3.3.3 with the Navio+ and I'm trying to connect to LightWare SF10 sensor as explained in the [wiki](http://copter.ardupilot.com/wiki/common-lightware-sf10-lidar/). Although I can see the device listed when it's connected to the I2C bus, I'm not getting any readings. 

pi@navio-rpi ~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- 
40: 40 -- 42 -- -- -- -- -- 48 -- -- -- -- -- -- -- 
50: 50 -- -- -- -- **55** -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- 65 -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- 77  

Is the wiki page missing any new step (e.g. change the default I2C address on the SF10/x; which is 0x55)?

marcmerlin, 2016-03-29:
I have no idea whether this is a bug or not, but usually issues are for confirmed issues. User help should be done on the forums. If a dev replies that this is a bug, then we can open a bug.
So, I'll resolve this for now, please post on http://discuss.ardupilot.org/ and if turns out to be a confirmed bug, you can re-open this.

rmackay9, 2016-03-30:
I can confirm (based on a number of user reports) that the LightWare range finders I2C are not working with Copter-3.3.3 at least on Pixhawk.  I believe it will work on master though.

squilter, 2016-04-17:
Can confirm that it works on master.

rmackay9, 2016-04-17:
@squilter, great, thanks for confirming that.

squilter, 2016-04-17:
Wait... I was confirming that this issue should remain closed - not that it should be reopened.

rmackay9, 2016-04-17:
Ok, but it doesn't work on Copter-3.3.. which is the version the issue is raised for.

rmackay9, 2016-08-29:
Ok, So I guess we're never going to go back and fix this for Copter-3.3.3.  It's included in master and Copter-3.4 though so I'm going to close this.

closed 2016-08-29 by rmackay9