ArduPilot/ardupilot #3717: GPS update rate to 10 Hz

Yups1337, 2016-03-07:
For some time now the Ublox M8 series of GNSS receivers is the standard for typical pixhawk configurations.
As it is able to send a update rate of 10Hz trough the UART, we should use it for better position control and logging at higher speeds. Also the geotagging of photos would gain accuracy.

To send the RAWX data at 10Hz speed (M8T only) the serial bus speed might be raised also.

Best regards
Yannick

WickedShell, 2016-03-07:
Right now the GPS driver doesn't allow you to raise the frequency of the timing, so that would need to be changed inside the GPS driver. Additionally there are several warnings related to not maintaining the appropriate update rate that would need to be reconfigured. These two are fairly trivial changes, the additional change is that I think the EKF would need some tweaks to handle the higher rates. (IE some gains etc, but I'd need to talk with @priseborough to get an answer as to which parts should be changed).

With regards to serial bus speed we already raise the baud rate for logging RAW data as the RAWX stuff from multi-gnss can saturate the 38400 link :) We might have to raise all the speeds again though, but that might not be a bad idea regardless.

priseborough, 2016-03-07:
The main change in the EKF is the time delay parameter EK2_GPS_DELAY should be adjusted because the GPs will have less processing delay. I would like to see the time delay for the GPS published by the GPS driver as it is difficult for users to find the optimum value of EK2_GPS_DELAY for themselves.

WickedShell, 2016-03-07:
That really relates with me getting #2850 done doesn't it?

kyunghwan, 2016-03-14:
@WickedShell 
what is GPS driver specifically?
Is the library in the arducopter code? or Is the part of gps module?
Thank you for answer

WickedShell, 2016-03-14:
It's common ardupilot code, its used to configure the gps module and
provide a interface to the data to be used by the rest of the code.

smr277, 2016-05-09:
What is the exact flies that need to be looked at? Plus, what is the status on this?

WickedShell, 2017-01-13:
This went in quite some time ago as #4844 but we never closed the issue.

closed 2017-01-13 by WickedShell