ArduPilot/ardupilot #3720: Plane: Cruise Mode Throttle/Pitch Behaviour Different to RTL

SlipstreamFPV, 2016-03-08:
On Arduplane 3.5 I noticed that the way the cruise throttle behaves is different to other auto modes such as loiter and RTL. The output differs as does the speed scaling. 

On RTL mode mid throttle = TRIM_ARSPD_CM, on Cruise mode mid throttle = Speed 1/2 way between  ARSPD_FBW_MAX and ARSPD_FBW_MIN.

Also cruise appears to be subject to more noise than RTL and Loiter throttle:

Cruise Pitch:
[image]

Cruise Throttle:
[image]

RTL Pitch:
[image]

RTL Throttle:
[image]

It would be good if all auto modes functioned the same regarding use of the throttle.

Using the users TRIM_THROTTLE setting as the middle point in all auto modes makes sense. TRIM_ARSPD_CM should be used as the speed the autopilot will fly at when the throttle is at this point. After the throttle has not been moved by the user in an auto mode for more than 2 seconds a deadband of 10/20us should be set around that point to prevent CRUISE adjusting the throttle due to small output dither from the controller Tx.

magicrub, 2016-03-08:
Do you fly with an air-speed sensor? ARSPD_ENABLE=1 and ARSPD_USE=1?

SlipstreamFPV, 2016-03-08:
Yes, I fly with the digital airspeed sensor and it has been tuned. The tuned value is very close to the default of 2.0.

SlipstreamFPV, 2016-03-08:
Just to clarify the "yes" means yes to all of your questions :) I can't edit the previous comment on my phone browser.

IamPete1, 2021-09-12:
@SlipstreamFPV is this still a issue on 4.1?

SlipstreamFPV, 2021-09-12:
I will test. I know there was another issue about this that is more recent, I will try and find the post.

peterbarker, 2022-09-16:
@SlipstreamFPV any update?

peterbarker, 2024-05-24:
Closing this as it's had NeedAnswer on it for years.

closed 2024-05-24 by peterbarker