ArduPilot/ardupilot #3723: Ardupilot controller - max climbing speed

gaucho1978, 2016-03-09:
We need to move the copter at 8m/s with an automatic flight program, but the maximum speed seems to be lower. is it correct? could we achieve a bigger value for the speed?
the parameter PILOT_VELZ_MAX accepts max 500 cm/s
i think that it is needed a parameter for the max descending speed and a parameter for the max ascending speed and the maximum value for this last one should be at least 2000 cm/s

magicrub, 2016-03-09:
Have you tried a larger value? Many of those limits are not enforced and any value would work.. assuming they're physically possible to achieve by your aircraft.

gaucho1978, 2016-03-09:
no, i didn't

rmackay9, 2016-03-10:
In autonomous modes (AUTO, Guided) the vehicle uses the WPNAV_SPEED_UP and WPNAV_SPEED_DN parametes.  As @magicrub says, the mission planner (and other ground stations) may limit them on some screens (like the extended tuning screen) but you can set them directly to anything you want on the Full Parameter List or Full Parameter Tree screens.
There are no upper limits placed within the flight code itself.  Heck, Leonard's copter was doing >30m/s autonomously 2 or 3 years ago at the AVC competition.

closed 2016-03-10 by rmackay9

gaucho1978, 2016-03-10:
Thank you. 
Dji has big limits.  ardupilot is still the best.

magicrub, 2016-03-10:
Randy, shall we change the limit to Mach 1?