ArduPilot/ardupilot #3724: Compass calibration: Provide feedback so usablility can be improved

DonLakeFlyer, 2016-03-10:
QGC hand holds the user through compass calibration in a way that:
* Is fairly easy for a user to follow
* Generates very good compass values every time
* Is less error prone

We've found that compass cal was a huge problem in getting users to understand what they should do. And also, keeping them from doing it wrong and getting cal values that aren't as good as they could be wasn't possible without providing a high level of visual feedback.

If you look at this video: https://www.youtube.com/watch?v=91VGmdSlbo4 and move to the 2:39 point you can see an older version of QGC doing compass cal. It is pretty close to how it works now. Future versions of QGC will even rotate a 3d model to show even better what is needed.

The way it works with QGC+PX4 is that the firmware is reponsible for doing the actual calibration, but sends back status message to QGC such that it can drive the ui. Here is the information that comes back from the firmware in the form of specially formatted mavlink status text messages:
* Which orientations have already been completed
* When the vehicle is stationary in one of the incomplete orientations. QGC then signals the use to start rotation and around that axis.
* A progress indicator for rotation around each orientation
* The progress indicator will continue until the firmware has enough good points to form the values for that specific orientation.
* When the orientation is complete it will tell QGC that. Then the whole process moves on to a new orientation.
* The user can choose which of the incomplete orientation they want to do next.

This process leads to repeatable excellent compass cal offsets.

For QGC+ArduPilot QGC simulates this process all on the ground station side by handling all of calibration itself. So you get the same ui, but QGC ends up in the calibration business which isn't a good thing. Also given the current support, there is no way for QGC to calibrate a third compass. This is due to MAV_CMD_PREFLIGHT_SET_SENSOR_OFFSETS only support two compasses, in combination with the fact that there is no way to save compass calibration values without using that command.

It would be great if ArduPilot could provide feedback similar to PX4 such that QGC could get out of the calibration business and also be able to calibrate all three compasses. 

FYI: In looking at the Mission Planner code you'll see that as far as I can tell MP is also not capable of calibrating three compass. If you look here: https://github.com/diydrones/MissionPlanner/blob/master/MagCalib.cs#L1178 you'll see that for the third compass it just sets params, which will not work if the eeprom device id for compass 3 is 0.

meee1, 2016-03-10:
please look into 
MAV_CMD.DO_START_MAG_CAL
and
MAVLINK_MSG_ID.MAG_CAL_PROGRESS
and
MAVLINK_MSG_ID.MAG_CAL_REPORT

what you have found is the old legacy code.

DonLakeFlyer, 2016-03-10:
Yeah I forgot about the new thing.

lucasdemarchi, 2016-03-10:
Problem here then seems to be:

 - We report only a "global" progress
 - We don't have a notion of "per-orientation" step (an hence the issue above).

DonLakeFlyer, 2016-03-10:
Yup, that's mostly it.

robustini, 2016-03-10:
In my opinion we should think about a much simpler calibration of the compass, DJI style.
Not everyone has small drones, calibrate the compass on this is like going to the gym... ;-)

[image]

lucasdemarchi, 2016-03-10:
I have never seen a dji calibration. But looking at this: http://wiki.dji.com/en/index.php/Phantom_3_Professional-Calibrating_the_Compass

  it looks very similar to the calibration process in QGC except that QGC asks for all directions instead of only 2. Doing one direction at a time IMO improves a lot the usability compared to the other methods.

jberaud, 2016-03-10:
@lucasdemarchi Try the compass calibration on your Bebop, I find it quite nice to use too.

lucasdemarchi, 2016-03-15:
@DonLakeFlyer We talked about this in the dev call today. The first thing to do is to start filling the bitmask of sphere sections: https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/ardupilotmega.xml#L1437  - see the linked algorithm, too. This would give the feedback in each direction. From what I see the axis-detection would still be in the GCS though, but we can figure that out once the bitmask is filled.  In order to have better values we may also need other directions like rotating the uav 45 degrees.

DonLakeFlyer, 2016-03-15:
I think I sort of get how I would determine whether a specific orientation is complete from that.

bugobliterator, 2016-04-05:
@lucasdemarchi that message is not implemented, @DonLakeFlyer The whole idea behind current ardupilot's compass calibration is to provide infallible and fast process. And it succeeds in both ways, to do an compass calibration, the user is required to fill the bar (which is a global progress). The decision behind choosing a global progress is that user is not required to rotate in all axis. As soon as samples collected are properly distributed, the compass calibration should stop. On the user end he/she should just rotate in random axes at whatever rate he/she can. The calibration usually takes no longer than 3 mins (on Solo), which is faster or at least as fast as existing calibration out there. Other thing its very difficult to fool the calibrator, the calibration will not succeed if you are in a magnetic field environment, which was not found to be the case during my comparison with PX4's compass calibration. Hence if APM's compass cal succeeds its definitely done in a very good environment and user might not want to do it for a very very long time. For solo, we don't even ask users to do compass cal after unboxing as this process is done on the factory line and is supposed to last until user puts the compass on a magnet, in that scenario the preflight check fail and user is told to do the calibration and its good to last until next time.

guludo, 2016-04-05:
@bugobliterator I'm doing some working on this - on filling that bitmask. Could you please take a look at https://groups.google.com/forum/#!topic/drones-discuss/RM4ekehnrO4 ?

I wanted to understand those formulae better...

Also, I'm assuming that bitmask is a tessellation of an icosahedron by dividing each triangular face into 4, yielding then 80 faces. Is that right?

DonLakeFlyer, 2016-04-05:
> The decision behind choosing a global progress is that user is not required to rotate in all axis. As soon as samples collected are properly distributed, the compass calibration should stop.

Have you put that in front of a decently large set of real users and have they been able to figure out what to do? The Solo forums seem to have many examples of confusion as to what to do. Not to mention all sorts of urban legend around things that are needed to do that are not true. Like pointing north and so forth. If you don't hand hold people will make things up. If people really are figuring this out then fine. But from my own poking around I'm seeing confusion.

DonLakeFlyer, 2016-04-05:
> If people really are figuring this out then fine.

Is this is the case, then I can switch QGC to shows a tumbling image like Solo does.

bugobliterator, 2016-04-06:
@DonLakeFlyer facing north while doing calibration is not a urban legend, its actually the right way to do calibration to rotate about the axes perpendicular to field vector(pointing south to north), which is easy to do when you are facing North/South. Anyways I agree that there needs to be more info (probably a video) on how to do calibration. But I have not heard any questions or confusions on this point. It would be gr8 if you could point me to the issues you found out, users are having.

DonLakeFlyer, 2016-04-06:
Let me do some digging for the issues I was talking about. I found then when I was trying to find out why my Solo wouldn't complete a compass cal. The issue with the Solo turned out to be the fact that the Solo seems to be very tempermental to completing compass cal indoors. Whereas most everthing else I have can seem to get good compass cal numbers indoors. I think I searched for "solo compass calibration failed" or something along those lines.

I still come back to the fact that if you need a video to show people how to do it, then why not hand hold them through a standard repeatable process that is easily understandable? It's the same as making someone watch a video but more directed. The reason QGC does it the way it does was to get around all the videos showing you what you need to do. Even all the videos that I've watched when I first started out are all slightly different making you wonder what is really right. Point north, don't really need to point north. Put it in a chair. Hold it out in front of you in various ways. Spin around in a specific way. PX4 Stack firmware based compass cal used to worked very similarly to the new onboard ArduPilot compass cal. With a progress indicator to tell you when you were done based on error rate I believe. But no help in the Ground Station app to tell you how to do it. It also relied on videos to explain to people how to do it and it ended up being a mess. Then I switched to the new UI and haven't heard much of a peep about users not understanding compass cal since then. We originally started with only asking for 3 orientations but switched to asking for 6 because it significnatly higher quality values.

lucasdemarchi, 2016-04-06:
> Anyways I agree that there needs to be more info (probably a video) on how to do calibration.

main problem that I see is that there's no feedback to the user on what is missing to complete the calibration. You can tell him to rotate in random directions and the directions will probably not be sufficient because they are biased to rotate in the same direction. My point is that if even us developers are having problems with the steps to complete the calibration, there's something wrong/incomplete with the steps.

Using the method from QGC or a similar one ( for example we could use a global percentage but "fill each direction that was completed") is very clear to the user on what he has to do. I think we can also finish the calibration earlier, before completing all directions if we have numbers good enough. This is not what I see in practice, though

DonLakeFlyer, 2016-04-06:
> I think we can also finish the calibration earlier, before completing all directions if we have numbers good enough.

Doable, but unsure if that would be better. Sometimes making things always be exactly the same is better. PX4 Stack calibration is 7 seconds per side  * 6 sides. So takes little time. The pain comes with large vehicles where side skipping would be helpful.

Naterater, 2020-11-11:
I think recent improvements to compass automatic orientation and new calibration options render this issue solved. Please reopen if not.

closed 2020-11-11 by Naterater