ArduPilot/ardupilot #3726: support for OpenPilot Revolution boards

night-ghost, 2016-03-10:
OpenPilot Revolution and Revolution Mini boards are quite modern and quite cheap, so it would like to be able to use them not only for Cleanflight but for Ardupilot too.

PR: https://github.com/diydrones/ardupilot/pull/3725

marcmerlin, 2016-03-29:
I see you are working on this with Tridge. I'll let you handle this issue and resolve it when the time comes.

night-ghost, 2016-04-05:
All pull requests are now closed, although the code in the first of them flew successfilly. It looks like ardupilot team is not interested in supporting this board

OXINARF, 2016-04-05:
@night-ghost It doesn't matter if it flew successfully if the code doesn't meet the standards of the project. If you want to submit a PR that meets these standards I'm sure the team don't have any problem in bringing it in. But don't expect them to do the work for you, they already have enough in their plate.

marcmerlin, 2016-04-05:
@night-ghost, I'm going to assume you're the same person doing the work on minimosd. If so, a big thanks for your great work you've done there with your fork, including so many more features while saving space.
I had a quick look at your pull requests, and from that I can see, you hit the typical issues in making big merges  with git and github, it's hard to get each set of changes in a clean independent pull request that can be reviewed by people with sometimes little free time as they work on a lot of code themselves. In some ways, with changes like yours, contributing to ardupilot can be almost as hard as contributing to the linux kernel.
All I can recommend is not to give up, be patient with git and github, and our developers. They do want your code, but it can be hard to get everything working together sometimes.
Your work with minimOSD has been excellent, I hope you can get your port rebased in bits that can be reviewed and merged.
I am going to close this issue though, simply because it's not needed. We don't need issues in addition to pull requests: your pull request is the master issue and this separate issue is just a dupe. I can however be helpful to someone looking for your port with google, in which case they will find this issue, even after I close it.
So again, please don't give up, even if I don't have your board, I would very much like to see some of your code in ardupilot too.

closed 2016-04-05 by marcmerlin