ArduPilot/ardupilot #3727: flight land final's flare climbrate update seems wrong

nocternal, 2016-03-10:
in  AP_TECS.cpp
it's update 50Hz,but in final land stage, flare's climrate change is simply counter 10 times ,is that meens the ture climbrate transition time is 0.2s,not 1s as it's comment said?reletive codes lists here:

if (_flight_stage == FLIGHT_LAND_FINAL) {
        _integ7_state = 0;
        if (_flare_counter == 0) {
            _hgt_rate_dem = _climb_rate;
            _land_hgt_dem = _hgt_dem_adj;
        }
        // bring it in over 1s to prevent overshoot
        if (_flare_counter < 10) {
            _hgt_rate_dem = _hgt_rate_dem * 0.8f - 0.2f * _land_sink;
            _flare_counter++;
        } else {
            _hgt_rate_dem = - _land_sink;
        }
        _land_hgt_dem += 0.1f * _hgt_rate_dem;
        _hgt_dem_adj = _land_hgt_dem;
    } else {
        _hgt_rate_dem = (_hgt_dem_adj - _hgt_dem_adj_last) / 0.1f;
        _flare_counter = 0;
    }

marcmerlin, 2016-03-12:
@tridge @magicrub any idea on this one?

magicrub, 2016-03-13:
Seems legit.

There is also a gremlin I've seen where the height demand goes to zero for one cycle at the flare. This section of code needs to be looked at closer. thanks @nocternal

magicrub, 2016-03-13:
turns out it's actually 10Hz. The comment is what's wrong.

update_alt @ 10Hz -> update_pitch_throttle -> _update_height_demand

magicrub, 2016-03-13:
ha! I don't even see a comment that says 50Hz. No problem here... closing.

closed 2016-03-13 by magicrub

nocternal, 2016-03-13:
thanks(^V^)