ArduPilot/ardupilot #3729: Plane Request: Track course with wings level using rudder.

iskess, 2016-03-10:
While mapping, the ailerons are working very hard to maintain the course which creates constant small adjustments to the bank angle. Depending on the camera settings and ambient conditions, this can create motion blur and requires a little extra sidelap to compensate for oblique imagery. In some circumstances it is ideal to capture the most nadir imagery possible, and gimbals in planes are very rare.  
When mapping in full sized aircraft, a strategy often used for small course corrections is use the rudder to bracket back on course while using the ailerons to keep wings level. This is also common during an ILS approach. 
I think a similar method could be used here, for small lateral course corrections between waypoints. When the course correction required exceeds a certain value, ailerons will need to be blended in, until a point where they become the primary control surface.

magicrub, 2016-03-10:
Limiting roll and pitch during a photo run is good for better photos but
bad for flight stability. Won't handle wind or hard turns well and therefor
whatever solution we come up with for reduced motion blur would need to be
temporary during the imaging pass and self-abortable to maintain a safe
flight.

proposed method:
- new mission item DO_REDUCED_MOVEMENT (needs better name) that you place
just after the start of an imaging run wp (between two imaging run waypoints
- limits roll/pitch to whatever degrees are in the mission item. 0 means no
limits (normal).
- resumes normal roll/pitch limits when reached next waypoint
- resumes normal roll/pitch limits when xtrack error gets above a certain
threshold (5m?) to handle situations where we are getting blown off course
and can't maintain.

Another solution is to trigger a special time constant or flight surface
dampener during this run section so everything just slows down, reducing
the blur.

iskess, 2016-03-10:
I was hoping to use the rudder as the primary lateral control surface during these times. 
I already have two CAM_TRIGG_DIST commands for every turn, this is going to get quite complex to add 2 more commands for every flight line. 
Since this would only be in effect during Auto flight when small course corrections are needed, what is the harm of making it an optional parameter?

magicrub, 2016-03-10:
it basically ruins the flight stability of the aircraft. Turns and wind tolerance are gone. You might as well just turn TECS_TIME_CONST = 10 which will give you a similar effect (to all flight surfaces). Or set PTCH2SRV_TCONST and RLL2SRV_TCONST high and STEER2SRV_TCONST low. It gives you the same effect but creates a lousy flight generally.

iskess, 2016-03-10:
But I'm looking at a conditional situation that is only active when less than, say 5m off track, or 10 degree correction commanded, with no WP course change. 
Your proposed work arounds would be active all the time and would certainly not be desirable.

WickedShell, 2016-03-10:
Driving at the moment, but what Iam is proposing is the best approach for
mapping if you have the surfaces available to do it. It should only be used
if you are close to being on the correct track/roll but should then kick in
for mapping.

magicrub, 2016-03-10:
Instead of me making stuff up, could you be as specific as possible with what and when and when not to modify which movements and/or flight surfaces.

Yups1337, 2016-03-11:
You could trigger the "reduced movement" with the "DO_SET_CAM_TRIGG_DIST" command. This is set to 0 during turns and =!0 during mapping. A fallback distance error is needed fo sure. (5m for example)

So basically there is a need for two new parameters: 
- reduction of aileron movement for navigation during mapping in percent [0-100] [default 0]
- distance error to navigation track in decimeters [0-255] for fallback to normal navigation mode 

(my suggestion)

iskess, 2016-03-11:
@Yups1337,
Just to make the distinction, I was not proposing the reduction of aileron movement, but rather the function of the ailerons. I propose that the ailerons act to keep the wings level if only small heading changes are required (<10 deg, an <5m course deviation), and the rudder functions to execute the heading change. 

Interesting idea to combine this function into DO_SET_CAM_TRIGG_DIST if activated by a parameter. This should serve to limit its use to the mapping phase of flight and not the repositioning or approach phases. Thanks for your idea.

iskess, 2017-01-13:
I'm wondering if there has been any further thought about this, or plans to implement it.