ArduPilot/ardupilot #3736: Traditional heli bad motor channel configured on failsafe

Fortesque0, 2016-03-11:
Hello,
The failsafe it´s looking for throttle on channel 3, but the throtle input it´s  Channel 8 for traditional helicopters. So the failsafe doesn´t work properly.

And how will work failsafe throttle ouput, looking for diferents RSC Throttle inputs (Ch8, Dissable or setpoint)??

You can see that throttle (cannel 8),its under failsafe set point, but pixhak its not on failsade mode. Channel 3 (Collective pitch on helicopter) its over failsafe point too.
[image]
Now, you can see that cannel 3 its under failsafe setpoint, and pixhawk goes to failsafe mode
[image]

The same happends if you are using Helicopter sitl . 
Thanks

R-Lefebvre, 2016-03-11:
This is how it is supposed to work.  Failsafe on Helicopter is on Channel 3, just as it is for Multirotors.

The use of the word "throttle" to describe Ch3 is just an unfortunate circumstance of the close connection between the multirotor code and the helicopter code.

Fortesque0, 2016-03-11:
Hello R-Lefebvre,

At first, thanks for your job, and all who makes this posible ;)
And the throttle control output, when failsafe it´s activated, goes to channel 8? I have been testing on SITL and the channel ouput that is assigned for throttle is ch3, no ch8. 
I will try this night using hardware to see how ch it´s activated for failsafe throtle control.
Thanks

Fortesque0, 2016-03-11:
I can set throttle receiver ouput ch3 to pixhawk ch8 imput, because i am using a pwm-ppm conversor and physically i changed wires. But, if i use a PPM or SBUS native output receiver, how can i change the throttle chanenel configuration to be channel 8?
Thanks

R-Lefebvre, 2016-03-11:
I'm not sure I"m understanding the question.  But you cannot change the Failsafe channel to Ch8.  It has to be on Ch3, which is Collective Pitch on helicopters.

Fortesque0, 2016-03-12:
I test failsafe (ch3) on pixhawk(using a servo for throttle output ch8).  Activating failsafe on my transmitter (all channel going to minimum), pixhawk enables failsafe, but throttle output (ch8) goes to minimum too.
Have you test it? 
Thanks

R-Lefebvre, 2016-03-12:
Oh, I see what you're saying.

I have tested this in the past.  But I'll check again.

Which code are you using?

The problem could be in your radio though?

Fortesque0, 2016-03-12:
I am using last firmware 3.3.3, but I suppose from your answers how failsafe works. I need to set up on my radio a failsafe value for channel 8 (throttle)?
I was thinking that on failsafe, pixwhawk takes the control of channel 8 independently that how mode is setted on heli throttle setup.
Sorry for my mistake, but could be a good idea to pixhawk get throttle control. Imagine that I have selected ch8 for throttle on heli setup, and my receivers crash flying. Pixhawks failsafe will take heli ciclyc control, but no throttles, crasing the heli.

Fortesque0, 2016-03-12:
And how about cross channels (ch3-ch8) using a native ppm or sbus receivers, for heli setup?  I can't do it through may radio channel assignment.
Thanks

Fortesque0, 2016-03-16:
Hello Robert,

What about configure throttle/colective/gain set channel on native s-bus-ppm receivers? And get the control of ch8 on failsafe, independtly of how it is configurated (ch8, none, value set)?
Thanks

R-Lefebvre, 2016-03-16:
I think you're asking if you can move the channels around so that failsafe is on Ch8?

You cannot at this point unfortunately.

I had tried to move in the direction of allowing fully flexible input and output channels assignments, but haven't been able to get the concept past the group.

Fortesque0, 2016-03-16:
No exactly. I am trying to explain two questions, (sorry but i speak very well spanish,no english;) 
1.- I understand the trigger for failsafe to be on Ch 3. If i have selected throttle setup at fixed value setpoint, the output for channe 8 on failsafe mode its the selected value.  If i have selected throttle setup  to ch8, i need to configure on my radio failsafe, a throttle value for ch8.
On wiki, the failsafe is described:

"When the failsafe will trigger

If enabled and set-up correctly the radio failsafe will trigger if:
•The pilot turns off the RC transmitter
•The vehicle travels outside of RC range (usually at around 500m ~ 700m)
•The receiver loses power (unlikely) 
•The wires connecting the receiver to the flight controller are broken (unlikely).  Note: with APM2 only the ch3 connection between receiver and flight controller will trigger the failsafe."

So if i have selected throttle setup to ch8, and "•The wires connecting the receiver to the flight controller are broken (unlikely)", the throttle output goes to 0.
I think its necesary to, on helifirmware, set a fixed value for ch8 output (independently of what mode of throttle we have selected on heli setup)  when failsafe its activated.

2. For heli throttle setup, i cross the channels  from receiver using a ppm converter
[image]

But if i use a native ppm output receiver, how i can cross the channels ???
[image]
Channel 3 will be always channel 3. I can´t cross phisically. I need to cross by soft.
This should be do from heli firmware.
I hope I've been able to express well. 
Thanks

Fortesque0, 2016-03-28:
Any solution for channel assigment??

kaangoksal, 2017-08-27:
Hi, today we crashed because of the same problem, in failsafe mode the channel 8 output is not preserved, even though it is on RTL because of no RC RECEIVER, rc8out goes to minimum. Is this supposed to be like this? How come the failsafe does not keep control the rc8out ?

ChristopherOlson, 2017-09-27:
I do not believe this to be a valid issue that requires any fix. With all helicopters radio failsafe is on Channel 3, which is collective pitch, but called "throttle" in the ground station software. Your radio must be configured to drop below the normal low value in event of loss of signal to trigger it. All other channels in your radio and receiver must be configured to hold last received value from the radio transmitter.

When triggered the radio failsafe will execute the action you have selected. If LAND or RTL is selected the Pixhawk controls the throttle fine and shuts down the engine upon completing the landing with no radio signal present.

Tested today with Copter 3.5.3 in AUTO, STABILIZE and LOITER flight modes. The radio failsafe functioned perfectly every time. The helicopter continued with the flight in AUTO, landed, shut down and disarmed upon completion of the flight. It immediately landed, shut down and disarmed in STABILIZE and LOITER.

I do not find any problems in the ArduCopter software for this issue. I believe the problem to be misconfiguration of the failsafe function in the RC radio, and perhaps the wiki should be updated to be more clear on proper setup of the radio failsafe for TradHeli.

kaangoksal, 2017-09-29:
@ChristopherOlson thank you for your feedback, we appreciate it.

bnsgeyer, 2018-09-20:
I agree with @ChristopherOlson on this.  I believe the only time this could be an issue is if you are using RSC Mode 1.  Again I agree with Chris that you must set your radio so the receiver holds last value on all other channels except channel 3.  that would ensure that if you are in mode 1 that you wouldn't lose the engine.   Channel 3 must be set to go low because that is how the receiver signals the flight controller that no transmitter is present.  I'm closing this issue unless there is something I'm missing

closed 2018-09-20 by bnsgeyer