ArduPilot/ardupilot #3737: Deep stall landing

Vladimir1974, 2016-03-11:
There are many aiframe in FPV and UAV world which can use deep stall landing procedure. Is it possible to realize this type of landing in APM code in AUTO mode? It whould be very welcome feature.

Cheers, Vladimir

magicrub, 2016-03-11:
How do you trigger it on these platforms now?

Vladimir1974, 2016-03-12:
At this time to carrying out this type of landing during planing mission i set altitude in LAND point above ground about 10...15m, during flight when airplane reaches the LAND point I change mode to MANUAL and switch on deflection of stabilizer up on 45 deg by mean of mixing signal in RC. When the landing point near take off point it's not a problem, but in case of RTL to rally point with following the landing sequence it's a problem, because this point maybe not in line of sight of operator.
Cheers, Vladimir

magicrub, 2017-03-16:
There is a new landing library where you can set LAND_TYPE to 2 for Deep Stall. The parma is not on the wiki yet but will be as soon as Plane 3.8 releases.
http://ardupilot.org/plane/docs/parameters.html#land-type-auto-landing-type

@WickedShell is hopefully adding some wiki info about it soon..

closed 2017-03-16 by magicrub

WickedShell, 2017-03-16:
LAND_TYPE 1

Closed #3737 <https://github.com/ArduPilot/ardupilot/issues/3737>.

<https://github.com/ArduPilot/ardupilot/issues/3737#event-1002946415>, or mute
the thread