ArduPilot/ardupilot #3742: More failsafes

jcooper21, 2016-03-13:
It would be nice to have updated failsafes in APM Plane such as an RTL option if high winds are detected mid flight. It would also be nice to have an RTL option if GPS signal starts deteriorates (RTL before the signal is completely gone). Safety First!

magicrub, 2016-03-13:
Can you define "starts to deteriorate"?

jcooper21, 2016-03-13:
For GPS sat count reducing below X sats and hdop increasing. Other manufacturers have implemented these failsafes nicely such as sensefly.

magicrub, 2016-03-13:
Have you looked at the AFS params?
http://plane.ardupilot.com/wiki/arduplane-parameters

see AFS_MAX_GPS_LOSS and AFS_GPS_LOSS. This kicks in if we've lost lock on
GPS for 3 seconds.

-TomP

jcooper21, 2016-03-13:
That isn't proactive. GPS is gone once that failsafe kicks in, a deteriorating signal would allow for safer failsafe instead of once GPS is gone.

jcooper21, 2016-03-13:
I forgot to mention motor failsafe. If the motor fails in flight is it possible to make the plane orientate itself to the home position and glide towards it? This operation would cancel if the motor started up again.

rmackay9, 2016-03-14:
I think it might be a decent idea to look at the copter GPS/EKF failsafe which triggers when the EKF's position and velocity "innovations" climb above 0.8.  I think this is a better predictor than the number of satellites or hdop value.

Naterater, 2019-01-07:
I think this multi-faceted issue should be closed in favor of the following issues:
https://github.com/ArduPilot/ardupilot/issues/4491
https://github.com/ArduPilot/ardupilot/issues/3456

The EKF failsafe issue on GPS could probably be re-created as its own issue if somebody wanted.

IamPete1, 2021-08-08:
This issue is covered by the existing issues linked above.

closed 2021-08-08 by IamPete1