ArduPilot/ardupilot #3753: Plane: flexible waypoint type

magicrub, 2016-03-15:
@CraigElder has some ideas on multiple waypoint type for auto.See page 2:
http://www.faa.gov/documentLibrary/media/Order/ND/8260.44A%20Chg1.pdf

 - [ ] Fly-by WP (normal pre-turn via L1 - equivalent to large WP_RADIUS like 20)
 - [ ] Fly-over WP (through it - equivalent of WP_RADIUS=1)
 - [ ] Fly-past WP (turn on outside turn)

This can be accomplished via unused params in mission item for NAV_WAYPOINT.

CraigElder, 2016-03-15:
[11:57:06 AM] Craig Elder: http://www.faa.gov/documentLibrary/media/Order/ND/8260.44A%20Chg1.pdf
[11:59:18 AM] Craig Elder: FLY-BY WAYPOINT- A fly-by waypoint requires the use of turn anticipation to avoid overshoot of the next flight segment.

FLY-OVER WAYPOINT- A fly-over waypoint precludes any turn until the waypoint is overflown and is followed by an intercept maneuver of the next flight segment.
[12:05:51 PM] Craig Elder: Fly-by (FB) Waypoint. A waypoint that requires the use of
turn anticipation to avoid overshooting the next flight
segment.
Fly-over (FO) Waypoint. A waypoint that precludes any turn
until the waypoint is overflown, and is followed by either
an intercept maneuver of the next flight segment or direct
flight to the next waypoint.

EShamaev, 2016-12-17:
`               <entry value="16" name="MAV_CMD_NAV_WAYPOINT">
                    <description>Navigate to MISSION.</description>
                    <param index="1">Hold time in decimal seconds. (ignored by fixed wing, time to stay at MISSION for rotary wing)</param>
                    <param index="2">Acceptance radius in meters (if the sphere with this radius is hit, the MISSION counts as reached)</param>
                    <param index="3">0 to pass through the WP, if > 0 radius in meters to pass by WP. Positive value for clockwise orbit, negative value for counter-clockwise orbit. Allows trajectory control.</param>
                    <param index="4">Desired yaw angle at MISSION (rotary wing)</param>
                    <param index="5">Latitude</param>
                    <param index="6">Longitude</param>
                    <param index="7">Altitude</param>
               </entry>`

EShamaev, 2016-12-17:
First change: MissionPlanner update
https://github.com/ArduPilot/MissionPlanner/pull/1434

EShamaev, 2016-12-17:
Any idea why in Mission_Command in AP_Mission only p1 is present and it is 16 bit?

EShamaev, 2016-12-17:
Acceptance radius in meters was already implemented.
Fly-over WP (through it - equivalent of WP_RADIUS=1)
Just set p2 = 1

EShamaev, 2016-12-17:
[image]

EShamaev, 2016-12-17:
https://github.com/ArduPilot/ardupilot/pull/5402

magicrub, 2016-12-17:
The image looks good, can you post the mission that generated that flight?

EShamaev, 2016-12-17:
Yes, I am fighting with github for Windows, spoiled whole repo.

Here is mission:

[waypoint_update.zip](https://github.com/ArduPilot/ardupilot/files/658949/waypoint_update.zip)

EShamaev, 2016-12-17:
https://github.com/ArduPilot/ardupilot/pull/5403

magicrub, 2016-12-17:
We really need to change the wiki to recommend using source tree instead of
the GitHub windows thing.

magicrub, 2017-01-07:
Mission planner issue to take advantage of this feature: https://github.com/ArduPilot/MissionPlanner/issues/1449

closed 2017-01-07 by magicrub

iskess, 2017-07-09:
It would be very helpful for visualization to have a marker placed on the flight plan path to indicate these WP parameters. For "Acc radius" is would show a small marker the set distance before the WP, and for "Pass by" it would show on the set distance after the WP. This should display on the Flight Data map as well as the Flight Plan map, so the operator can know when the plane is expected to initiate its turn.  There could even be an indication on the HUD next to the WP Dist that shows the offset of when the turn will start. + means "Pass by" distance, and - means "Acc radius" distance.  For example, 500m to WP2 with a 'Pass by" of 15 would look like:  **2>500+15**