ArduPilot/ardupilot #3754: GPS: dual GPS switches between each other too easily

magicrub, 2016-03-15:
The current primary to secondary switch logic prefers the GPS with the higher sat count in an effort to gain better precision. However, that causes excessive switches which cause excessive position jumps. Its better to change the switch logic for dial GPS to have the second GPS to act as a backup for when one fails or degrades a lot, then do a switch.

The best solution is dual EKF instances running dedicated GPS instances described in issue https://github.com/diydrones/ardupilot/issues/3547

There's no ETA on that so lets discuss a better switching logic to replace the current one which is here https://github.com/diydrones/ardupilot/blob/master/libraries/AP_GPS/AP_GPS.cpp#L423

magicrub, 2016-03-16:
how about:
only swap if the other has 3 or more sats AND you are currently <= 7 sats

WickedShell, 2016-03-16:
One of the most common problems I've seen is people having one GPS that
struggle to maintains SBAS. Which causes it to flick between fix types 3
and 4, which whichever GPS has a higher fix type has priority. It would
maybe meaningful to limit how rapidly we will swap on those situations as
well.

Anything here just seems like it would be messing around without addressing
the real problem, and I think it will just represent a tradeoff, where it
makes the switching algorithm worse for some. (Although the proposed change
does sound fine). I'm not sure how hard it will be to get pauls time for
the EKF fix, but I'd like to champion that as something that deserves a
fairly high priority.

proficnc, 2016-03-16:
I second that... Pauls solution is the correct way forward here.

jpkh, 2016-03-16:
SBAS is not available on everywhere. Pauls solution is best so far

WickedShell, 2016-03-16:
Oh it's by far the best. The only reason I mentioned SBAS was thats one of
the most common swap situations to see that causes problems (in NA/SBAS
regions at least) and is relevant to tom's use case.

magicrub, 2016-03-16:
Switches cause EKF and position problems therefore should be avoided unless absolutely necessary. After much internal deliberation of looking at logs, Hao and I have decided that GPS2 (secondary) should really be treated as a backup instead as an additional sensor to increase performance at the higher GPS driver level. This is, of course, until the proper EKF integration happens where you get the best of both worlds. Therefore, switches should only be made if the primary GPS looses 3D_fix lock, irregardless of SBAS/WAAS or sat counts. We have many logs where the GPS drops to nothing and that is the cause for a switch. We have many other logs with SBAS/WAAS going in and out and that is not cause for a GPS switch.

Thus, we propose the switch should *ONLY* occur if:
 (GPS[pri].status > GPS[sec].status) AND (GPS[pri].status < 3D_fix)

This covers the case of GPS sensors dropping out and disregards the case where one may be slightly more accurate than the other.

Discuss.

magicrub, 2016-03-16:
Whoops, had a sign typo (fixed above). And actually this is better to show intent by showing 3D first:
(GPS[pri].status < 3D_fix) AND (GPS[sec].status > GPS[pri].status)

magicrub, 2016-05-18:
pending PR: https://github.com/ArduPilot/ardupilot/pull/4155

magicrub, 2017-03-08:
will be fixed by https://github.com/ArduPilot/ardupilot/pull/5642

WickedShell, 2017-03-20:
Blending went in with #5841, closing this as thats the better approach then hard swaps.

closed 2017-03-20 by WickedShell