ArduPilot/ardupilot #3766: Plane: Split up stream rate SRx_EXTRA3 by adding SRx_EXTRA4

magicrub, 2016-03-17:
There are way too many things in stream rate Extra3. See here:
https://github.com/diydrones/ardupilot/blob/master/ArduPlane/GCS_Mavlink.cpp#L1082

It's time we add a SRx_EXTRA4 and migrate a few extra3's over to it.

billbonney, 2016-03-22:
Why too many? Not all the message are sent all of the time. i.e. the CAL messages.

The only need to move them into a separate config is if you wanted to be able to set a different send frequency. Are some of them being sent too often? Do they need to be at a different rate.

The other challenge is that if you change it how to manage a transition. i.e. some message may now not be sent to a GCS when in connects unless it request EXTRA4 message sending.

To be honest having a system where each message has a request rate independent of these arbitrary groups would be much better.

Hope that helps :)

magicrub, 2016-03-22:
>>a system where each message has a request rate independent of these arbitrary groups 
Yes, that is the best solution but 15 or so SR params is not practical so major changes will be needed. It just seemed like Extra3 has become the kitchen-sink of streamrates.

magicrub, 2016-06-16:
Sorry @billbonney I misinterpreted your comment. We, of course, would not have 15 SRs. The GCS would subscribe to a mavlink packet at a specifiedrate. Each message would have it's own interval.

OXINARF, 2017-06-04:
Is this something we still want to do? This involves changes to MAVLink spec, changes to GCS and somehow have a transition method. EXTRA3 is also pretty full in Copter, but I don't think there is a good way to get out of it - except supporting the rate per message method.

magicrub, 2017-06-04:
Implementing rate per message is a better use of our time

closed 2017-06-04 by magicrub