ArduPilot/ardupilot #3767: download failed

jackson-xi, 2016-03-18:
[image]
Hi, everyone.Actually, I can't download the APM3.2.1 code successfully into APM2.6. I promiss I do anything accorading to http://dev.ardupilot.com/wiki/building-ardupilot-with-arduino-windows/. But the APM board can get the right appearance when I have downloaded the code. I will appreciate anyting.

magicrub, 2016-03-18:
I'be never built for arduino but it is considered legacy and has not been
supported by the main master branch for a while. You must use the
master-AVR branch.

rmackay9, 2016-03-19:
I think it should work with master if you go back to the ArduCopter-3.2.1 tag (or whatever it's called).  I think what's happening is that after compiling it's over the AVR's flash size limit.  It was really barely fitting and the code size can change a bit depending upon the compiler.  So if you use a slightly older compiler than we did it likely won't fit.
I'm going to close this because we don't support the AVR anymore, sorry!

closed 2016-03-19 by rmackay9