ArduPilot/ardupilot #3768: allow manual throttle in Position Hold

RQRobert, 2016-03-19:
Previously, Position mode's maual throttle was very useful for me  while maintaining attitude . 

But now with Position Hold i can't achieve it with constant attitude, so now i am moving back to AC3.1.5.

Kindly add manual throttle option (or similar) in PosHold mode so i can enjoy the very trustworthy EKF algorithm.

squilter, 2016-03-19:
I suspect most people prefer it this way.  I know that I do.  You can increase `PILOT_ACCEL_Z` for snappier altitude response.

R-Lefebvre, 2016-03-19:
Yes, I have a 250 Racer with 1G Pilot_accel_z, and flying it around in Alt Hold, it's hard to tell that it's not manual throttle, except I don't have to play with the throttle stick to hold altitude while zipping around.

RQRobert, 2016-03-20:
I am already using PILOT_ACCEL_Z param with PosHold but the results are not satisfactory .
First of all the copter starts climbing above 50% THR to 80% THR , second 
the max limit of Pilot_accel_z is 5m/s so it uses 100% THR stick and i can get only 3 m/s climb rate maximum
.
 To add , in order to descend copter down i have to lower throttle to 30-40%.

Once i was bringing copter down in poshold, crash_check triggers due to 30deg lean angle difference [in actual the descent was not smooth] resulting in a crash.

In a nutshell, Poshold is not an alternative of Position mode, kindly look into it.

Naterater, 2019-01-07:
Is this still an issue @RQRobert?