ArduPilot/ardupilot #3771: Feature request: Log any param changes into DF

hsteinhaus, 2016-03-19:
For reviewing tuning flights it is essential to know about parameter changes and their exact time. The params seem to get logged when tuned by knob (Log_Write_Parameter_Tuning() ), but there is no equivalent for manual param changes via MavLink telemetry. The same applies for parameter changes made by (Copter) Autotune.

The implementation could be easily done e.g. by writing a PARM message on change like already done on startup.

R-Lefebvre, 2016-03-19:
Strange, I was pretty sure this is already happening?

hsteinhaus, 2016-03-19:
Did definitely not happen in my flight three hours ago with latest master and LOG_BITMASK=180221. I changed at least three parameters during flight, only PARM messages in the DF log are from startup.

magicrub, 2016-03-19:
Convert log to .log and you should see them.

hsteinhaus, 2016-03-19:
Did that (via mavlogdump.py) and grepped for PARM, did not found any except on startup. Just uploading the log, so you can try as well...

hsteinhaus, 2016-03-19:
https://www.dropbox.com/s/hgu7iohv8bqyr6n/59.BIN?dl=0

What I would expect to see: about 3-5 changes of ATC_RATE_FF_ENAB (1->0->1->0->1), multiple changes on ACELL_Z_P, and two complete sets of PIDs (about 18 parameters in total) saved after AutoTune (run two times in that flight) completed.

hsteinhaus, 2016-03-19:
Changing the subject of the issue, as I found in a bench test that AP is generally logging the param changes, but...

AP is not logging them during disarmed state when MASK_LOG_WHEN_DISARMED is not set. Unfortunately AutoTune saves params on disarming, but obviously after the disarm happened. 

I still think that this behaviour is probably not what the user expects when disabling "logging when disarmed".

hsteinhaus, 2016-03-21:
I think this issue mixes up two different things, closing it in favor of #3778 and #3779

closed 2016-03-21 by hsteinhaus