ArduPilot/ardupilot #3774: Aileron reserve in flaperon config

rscarawa, 2016-03-20:
I have a Telemaster which I was using dedicated flaps and ailerons.   To help get a steeper approach on final, I wanted to induce more drag via crow.  With crow using the flaperon config, it brought light a limitation of the flaperon config and how it limits aileron movement.  When flaperon is turned on, it halves the movemenet in the ailerons that I used to have.    I believe this is to leave movement in reserve for the flaps and aileron  movement at the same time without oversubscribing the movement of the aileron servos.  The problem is I need all the movement on the ailerons that I can get for auto navigation purposes.  I am trying to fix some of this by changing the physical linkages to get more movement.  That has downsides in that any linkage slop will be amplified and the torque demand of the servo will be increased.

I cannot see anything in the Mission planner or docs that allows you to change the aileron or flap reserve on the flaperons.  It seems that Mission Planner is splits the aileron movement in half by default.  It would be nice to have control over how the aileron servo movement is allocated so I can control how much aileron I will lose in the flaperon config.  Maybe leave 70% for aileron and 30% for flaps on the aileron servos.  Maybe even allow the user to oversubscribe it if they choose.  Maybe 75% for aileron and 50% for flaps.  Any overage would just result in no additional servo movement.  This is common in many tx setups.  Should you need extra roll authority on final, you either add a little rudder or backout some crow.

Any thoughts?  Is this already possible and I missed the config somewhere?

Scot

rscarawa, 2016-03-21:
Going to see if the mixing gain will address my issue.

rscarawa, 2016-03-21:
For the mean time, Mixing_gain will do the job.  It would be nice though to be able to control how the flaperon movement is rationed between aileron work and flap work.

marcmerlin, 2016-03-29:
Mixing gain is indeed the way to go. I'd set it to 1
More generally, please don't use the issue tracker for general questions and help, but use http://discuss.ardupilot.org/ instead. 
As for doing full crow flaps with ardupilot, I don't think this is really well supported. I personally do crow flaps in manual mode for landing. If you'd like to open a forum post about it, please do so and post the link to it here, I'll subscribe since I'm interested too.
Also, see http://ardupilot.org/plane/docs/flaperons-on-plane.html if you haven't read it yet, although it sounds like you're already past that

closed 2016-03-29 by marcmerlin