ArduPilot/ardupilot #3778: Param changes not logged when vehicle is disarmed

hsteinhaus, 2016-03-21:
Currently, parameter changes on the disarmed vehicle are only logged if MASK_LOG_WHEN_DISARMED is set, otherwise they are suppressed. I think the intention of the MASK_LOG_WHEN_DISARMED flag is to suppress useless data streaming in from sensors and flight control, while the vehicle is sitting on the ground and waiting for action. However, the current implementation also suppresses logging of param changes that happen while the vehicle is landed (disarmed). 

This behavior is very nasty when tuning a vehicle and doing multiple flights battery, as parameter changes are impossible to trace after the flight.

marcmerlin, 2016-03-29:
Sounds like a valid concern

rmackay9, 2016-03-30:
There was a suggestion from someone (@jschall?) to add a option to start logging when armed but then keep logging even if the vehicle is subsequently disarmed.  Another option would be to rewrite the latest parameter values each time the vehicle is armed... I suppose it would rewrite the parameters only if some had been changed since it was disarmed.. although that would require some extra complexity in that we'd need to record when the last parameter value was updated and when the vehicle was last disarmed.

hsteinhaus, 2016-03-30:
@rmackay9 
I think to suppress the logging of mass data while the vehicle is disarmed is a very useful feature, especially in the context of logging hundreds of megabytes of mass data per hour. 

What do you think about #3870? It fixes the issue for my needs and did not show any negative impact during the last 5 or 6 hours of flying. @peterbarker has also presented an idea how to integrate this feature with log rollover on disarm (#3067). 

Generally, I think the problem is not limited to parameters, suppressing events and errors in the disarmed state could be the same story.  #3870 could be easily extended to cover this as well.

rmackay9, 2016-03-30:
ok, so if was just events, errors and parameter changes that were logged while the vehicle is disarmed it would be ok.  Just to be clear though, it wouldn't do any logging unless the vehicle had been armed at least once.

peterbarker, 2017-12-19:
This was fixed by logging critical messages when disarmed (https://github.com/ArduPilot/ardupilot/pull/7130)

closed 2017-12-19 by peterbarker