ArduPilot/ardupilot #3779: AutoTune does not log result values

hsteinhaus, 2016-03-21:
AutoTune uses AC_PID::save_gains() to permanently store parameter values in EEPROM. However, the final results are not logged directly. When doing more then one AutoTune per flight/battery, it is very hard to trace the results.

lthall, 2016-03-22:
Autotune logs the pid after each test so the final value are the final entry in the log for a given axis. 

This issue can be closed.

hsteinhaus, 2016-03-22:
Yes, this information can be reconstructed by looking on the latest value of AUTOTUNE.Gain_* at the point right before AUTOTUNE.Axis switches over to the next value (or doesn't if it was the last axis to be tuned). However, it needs a deep understanding of AutoTune to find out the actual PID values used for for a certain segment of flight. This e.g. prevents easy AutoTune result processing by scripts. 

Wouldn't it be possible to write a PARM message in the same way as done after manually changing PIDs by GCS? IMHO it would be a big improvement when every parameter written to the EEPROM would be logged under any circumstances.

closed 2018-01-11 by lthall