ArduPilot/ardupilot #3781: Improvement Waf px4 upload

khancyr, 2016-03-21:
Hi,

I just test waf build for px4-v2 (on Ubuntu 15.10 with cmake 3.2.2 , official toolchain and ninja) and it works pretty well (failed at start but after clean and dist-clean, it works and I couldn't reproduce the bug...) 

I test waf rover upload. It works !
But we are missing documentation in readme about upload . I used waf rover --upload

Here is what is done with make px4-v2-upload when usb cable is not plug : 
````
%% Generating /home/khancyr/Workspace/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.px4
make[3]: Entering directory '/home/khancyr/Workspace/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build'
Loaded firmware for 9,0, size: 880988 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
````

Here is waf version : 
````
[3/4] cd /home/khancyr/Workspace/ardup.../dsdl/uavcan -Oinclude/dsdlc_generated
Running from the source directory
[4/4] uploading /home/khancyr/Workspac...firmware/nuttx/nuttx-px4fmu-v2-apm.px4
````

So we don't know what happening ...  Some log/message/progess bar should be good ! 

After pluggin usb we got : 
````
[4/4] uploading /home/khancyr/Workspac...firmware/nuttx/nuttx-px4fmu-v2-apm.px4
Loaded firmware for 9,0, size: 902696 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
attempting reboot on /dev/serial/by-id/usb-3D_Robotics_PX4_FMU_v2.x_0-if00...
if the board does not respond, unplug and re-plug the USB connector.
Found board 9,0 bootloader rev 4 on /dev/serial/by-id/usb-3D_Robotics_PX4_BL_FMU_v2.x_0-if00
50583400 00ac2600 00100000 00ffffff ffffffff ffffffff ffffffff ffffffff 66ed47ff ff73cc15 c8ad940c dbc59f39 d6c20e06 f953d3ef f3073019 d035ab0d 3f60334e 10dda9f8 cdb0cbbd 42cdc6b6 3ba305f7 81532581 84ee3da6 23bc6340 8321be68 edd356c9 1e3b8f5c 5e07decc 9c6be5a2 458a1513 4bbbbc21 eda35ce5 a8b840a5 ef019ca5 c89bb183 bb00f0c0 06db1a26 7375ff57 1ca41d94 24aa662e ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff type: PX4Waf: Leaving directory `/home/khancyr/Workspace/ardupilot/build/px4-v2'
Build commands will be stored in build/px4-v2/compile_commands.json
'rover' finished successfully (18.827s)
````
So its good ! (Great work by the way ! ;-) )

lucasdemarchi, 2016-03-23:
@guludo This seems to be one more of those in which we are caching the cmake output when we shouldn't... ?

lucasdemarchi, 2016-03-23:
wrong button... should be "comment", not "close".

guludo, 2016-06-14:
Cmake tasks don't cache the command output for a significant time now. @khancyr, is that problem solved for you?

khancyr, 2016-06-14:
Yep, you can close. I give a try and now we got the right message : 
````
[2/2] uploading /home/khancyr/Workspac...firmware/nuttx/nuttx-px4fmu-v2-apm.px4
Loaded firmware for 9,0, size: 920972 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
````

closed 2016-06-14 by khancyr