ArduPilot/ardupilot #3782: Bug found, arming_check not fullfilled, can cause flyaway

furynick, 2016-03-21:
I had a crash this afternoon when battery failsafe was engaged.
After reviewing the log it appears I successfully armed Copter with HDoP of 100, ARMING_CHECK=1 and GPS_HDOP_GOOD=160 (might prevent arming if HDoP greater than 1.6).

I [uploaded log on Exmaps](https://www.exmaps.com/8b7976).

My setup is quite simple : F450 frame, AUAV-X2, external GPS/Mag, X2212/900KV motors, 10x4.5 prop, 4S/3000 lipo.

squilter, 2016-03-21:
Lower HDoP is better.  I'm not sure what the problem is.

rmackay9, 2016-03-21:
Github is really for confirmed bugs and enhancement request, not support calls so I'm going to close this for now at least.  Sorry to be strict but if we're not it will just turn into a support forum.  I know it can be hard to get support on issues but if we let support overwhelm the developers we will never get another release out.

The exmaps page is very impressive, i haven't seen it in a long time.  Sadly I couldn't find a way to download the .bin file.  I'd prefer to not create an exmap login at this time so can you put the .bin file somewhere. 

While I'm closing this item I'll be notified of replies.

closed 2016-03-21 by rmackay9

furynick, 2016-03-22:
@squilter : the HDoP was 99.99 when I armed, it comes back lower to 1.6 afterwards.
@rmackay9 : so sorry but I'm not aware of the good way to submit a support call so I understand :).
Anyway, the .bin log file can be downloaded at this url :
https://drive.google.com/open?id=0ByipkHiiI39KS2FsRnozZkZrdzA
Moreover, if I'm right (RTL to 42.5584268, 43.5985443 instead of 43.5350771, 1.4130001) the copter goes the wrong way (goes to West instead going to East, the shorter way).

rmackay9, 2016-03-22:
@furynick,
Ok, so it looks like the vibration levels are *really* high on this vehicle.  My IRIS has vibration levels below 15m/s/s with occasional peaks to 30m/s/s.  60m/s/s is the recommended maximum but in this log they appear to be regularly over 100m/s/s.
To help catch this, the Mission Planner will show "VIBE" in red on the HUD when levels are this high.  I think some of the other GCSs will too including Tower but perhaps only if enabled through one of the menus (I'm not sure on this point)

BTW, If the vehicle takes-off in a manual mode without GPS, then RTL won't know exactly where to return to.  It will attempt to return to the location at which it first got a good stable GPS lock.  In general I always switch to Loiter and wait for the LED to go green to ensure that I have a good home position recorded for RTL.  I'm  not sure if this is related to the vehicle's bad RTL behaviour.. from the NTUN log we can see that it was having trouble controlling it's velocity which is more often an indication that the EKF's position and velocity estimates were off (because of the high vibration levels).

We have wiki pages re vibration here:
http://copter.ardupilot.com/wiki/common-measuring-vibration/
http://copter.ardupilot.com/wiki/common-vibration-damping/

furynick, 2016-03-22:
Thanks for this answer, I'm aware of vibrations due to props and DJI arms and causing unstable AltHold.
After checking another time the log it's my mistake, I disabled the fence at some time and didn't set it back to on.

Nevertheless, perhaps this situation can be avoided by disabling RTL when home position can't be reliably stored (HDop > GPS_HDOP_GOOD when vehicule is armed).

rmackay9, 2016-03-22:
@furynick, there's an issue on the to-do list regarding this including some discussion about what the best thing to do is: https://github.com/diydrones/ardupilot/issues/910

I'm not sure completely disabling RTL is a good idea.  I would prefer the ground station allows the user to set the home location manually while the vehicle is in the air.  The Ardupilot side of this is already complete so it's more about GCS implementation.

It's quite difficult to make everyone happy I'm afraid.  People want to be able to fly without having GPS, they also want RTL to work.  The desires are somewhat incompatible and so I fear it will be difficult to make everyone happy.

furynick, 2016-03-22:
Going on #910 to continue, I think there's a softer way to have everyone happy and safe.
Many thanks for your help.

furynick, 2016-03-24:
I come back on this issue after reminding an important detail.
During my flight I landed, the copter autmatically disarmed and I armed it again before takeoff, why didn't it store home position at this time ?