ArduPilot/ardupilot #3783: servos quiet on boot using DSM input - rare event

magicrub, 2016-03-21:
I've only seen this a couple times so it' shard to catch. When booting with a DSM satellite RC receiver and the transmitter turned off, I sometimes get no output on my servos. Power flags look good (3), Vservo is stable at 5.2V and channel in/out all look good on ArduPilot telemetry. Just a flatline 0V on the servo rail. Reboot fixes it.

WickedShell, 2016-05-16:
Reproduced the other day on ArduPlane V3.5.2 (22e94dbb). Satellite was reporting that it had link, but no servo movement. RCIN and RCOU do show movement commanded so I'm not sure why nothing happened there. I forget the status of the IO LED at this point. Using the IO override did not grant control either. Log is available here: http://files.swiftradioplanes.com/27.BIN

magicrub, 2016-05-16:
FINALLY SOMEONE ELSE SEEMS THIS! Thank you @WickedShell . Now, if only we had any clues to what is actually causing the problem. The only thing that really comes to mind is a race condition in the px4io serial rx handler that depends on the timing of the first DSM parsed message. Only a guess though.

magicrub, 2016-05-27:
fixed by sending repeated attempts until the armed state is correct: https://github.com/ArduPilot/ardupilot/commit/67aaf7e226fd1ac9d6acd3589a53e962dab91588

closed 2016-05-27 by magicrub